ARMmbed/mbed-os #4300: STM32F4 SPI driver return error causing SD card driver to fail

LiyouZhou, 2017-05-11:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2 and NUCLEO_F429ZI

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3 20150529 (release)

**mbed-cli version:** 
1.0

**mbed-os sha:**
42be5c01a7f91292d5e27124ad9584236025f6ab

**DAPLink version:**
0221

**Expected behaviors**
SPI driver should be thread safe

**Actual behavior**
In a multithread environment. SPI driver returns TIMEOUT error from time to time, which the SD card driver is not handling, causing filesystem corruption.

I'm not sure if the spi driver is suppose to return -1 as error code because error code is not defined in the api.

Also I got a trace of the SPI signals using a logic analyser, all looks ok. Hence the error could only originate from the software.

The error originates from this line https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_spi.c#L964 which is an HAL_TIMEOUT error.

I think when the thread running spi gets swapped out and back in again while receiving, the timeout check fails.

When I increase the timeout to 50ms from https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/stm_spi_api.c#L360, the problem seem to go away.

0xc0170, 2017-05-11:
cc @bcostm @adustm @LMESTM @jeromecoutant

JanneKiiskila, 2017-05-11:
NOTE! This is super critical to get fixed ASAP - we can't get the mbed Cloud Client work reliably with these boards unless this gets fixed.
@0xc0170 @geky @bridadan @adbridge

LMESTM, 2017-05-11:
@LiyouZhou  thanks for the detailed analysis. 

So let me see if I understand correctly: 
- the SPI transfer is being delayed because of other ongoing threads having higher priority, right ? 
- because of this delay the SPI driver considers hitting a timeout issue -> we can of course increase this timeout to an "hardcoded" minimum value - but then what would be an acceptable value, is 50ms realisitic, or are there cases where the other threads can delay the processing even further ? 

In the meanwhile, considering the urgency, please feel free to send a PR using 50ms timeout instead of 10ms.

0xc0170, 2017-05-11:
@LiyouZhou Please send a PR with a proposed fix.

> SPI driver should be thread safe

It should be. @LMESTM why it was 10ms set previously as a default value? Is there any critical section within spi driver? How does this timeout work internally? what if there's is an interrupt during the transfer?

cc @andreaslarssonublox @andreaspeterssonublox

marcuschangarm, 2017-05-11:
Should there even be a timeout set in the first place? 

The function in question is an implementation of the mbed-hal. Having a timeout on some boards but not others would result in unexpected behavior for people using mbed OS for its portability.

LiyouZhou, 2017-05-11:
https://github.com/ARMmbed/mbed-os/blob/master/hal/spi_api.h#L117 the mbed hal doesn't define errors.

Problem is I couldn't say for certain how long a delay is enough.

LMESTM, 2017-05-11:
@LiyouZhou : 10ms was set as it was considered an acceptable value for sending and then receiving 1 byte in a frequency range from few 100s kHz to few MHz.  At 100kHz, which is about the lowest frequency available on most stm32 targets, the transfer is supposed to last around 200us, so 10ms sounded like a safe  value as this is 50 times more than supposed worst case, but I'd be happy to increase it further no problem. I can also post the PR if you prefer so. 

The timeout is based on stm32 HAL tick which is incremented under interrupts. 

One more question as I'm actually interested in understanding the whole system: as I understand it, there are threads with high priority that monopolize full CPU usage during 10ms, aren't there risks for other real-time tasks than SPI to fail ?  Can you share what the thread with higher priority is in charge of ?

LMESTM, 2017-05-11:
> https://github.com/ARMmbed/mbed-os/blob/master/hal/spi_api.h#L117 the mbed hal doesn't define errors.
Indeed, so what are we supposed to do in case of error ? return a default value (0?), call for erro() and die ? 

> Problem is I couldn't say for certain how long a delay is enough.
That's why I'm asking more questions in above comment, I'd like to better understand the situation.

LMESTM, 2017-05-11:
I have sent the PR for anyone willing to unblock the situation. Nevertheless I still would like to better understand what happens in the system that cause the transfer to last more than 10ms. I ten to think that an alternative might be to use high priority threads for low level HW communication, and ensure that higher level application SW has lower priority. But maybe this is already the case and the issue is different. I may need a way to reproduce the issue on my own to help further.

marcuschangarm, 2017-05-11:
@LMESTM The problem we are facing is that no one entity is in charge of the entire application. What would be considered high priority for one party might be considered low priority for another. And even with all high priority threads, we would have delays caused by the scheduler round-robining between threads of equal priority.

adbridge, 2017-05-11:
@marcuschangarm Do you think the PR will unblock you sufficiently for now? With the caveat that this is a more complex issue which needs further consideration?

LMESTM, 2017-05-12:
@marcuschangarm even with many threads and Round Robin, I'd like to understand what the system is actually doing that makes the SPI driver not able to complete the transaction within 10ms. the CPU is running @80MHz and that makes a lot of CPU cycles ... just would like that you check what actually happens and make sure there is no active-waiting-loop somewhere in another driver, or maybe consider using SPI trasnfer by blocks rather than 1 byte at a time ....

marcuschangarm, 2017-05-12:
@adbridge Yes, this should unblock us so we can continue development. We would need to have something more robust in place before the public release of our new product.

@LMESTM Thank you for the quick turn around! The weird thing is, there shouldn't be any heavy threads running when this error occur. However, I have little control over what threads in other libraries are doing.

LMESTM, 2017-05-12:
@marcuschangarm I think that your case deserves a good profiling session to understand what happens which leads to this situation. maybe there is a bug in my driver, but mightbe as well a bug elsewhere that has first side effects on SPI

JanneKiiskila, 2017-05-12:
I think mbed OS would greatly benefit from having a test case that would do in parallel

1) SD-card file reading/writing
2) HTTPS download (over async API).

That would reveal majority of this kind of issues.

marcuschangarm, 2017-05-12:
@JanneKiiskila I agree tests like that would be useful for all mbed OS drivers and we should ensure that different peripherals doesn't interfere with each other.

However, in this particular case the SPI timeout happens after we are done downloading.

JanneKiiskila, 2017-05-15:
So, what's the best plan for getting this into a mbed OS patch release? We would need it really soon to enable more boards for the next mbed Cloud Client release.

0xc0170, 2017-05-15:
> So, what's the best plan for getting this into a mbed OS patch release? We would need it really soon to enable more boards for the next mbed Cloud Client release.

Should get to the next one, shall be reviewed today during gatekeepers sync

LMESTM, 2017-06-02:
@adbridge @LiyouZhou  should this be closed now ?

marcuschangarm, 2017-06-05:
I'm happy to consider this resolved.

adbridge, 2017-06-05:
@JanneKiiskila Are you also happy to close this now ?

LMESTM, 2017-06-09:
ST_TO_BE_CLOSED

closed 2017-06-09 by 0xc0170