ARMmbed/mbed-os #4304: writing new target for my new custom board

YarivCol, 2017-05-11:
Hello , 

Im tyring to write a new mbed target to the my custom build board,
My board has a ST mcu and while trying run it i noticed that the HSE_VALUE defined inside the hal_conf file is shared between every STM32 target.
My question is, if ill connect another STM32 PCB that will need other HSE_VALUE how do i do that as there is only one file defining the HSE_VALUE.

0xc0170, 2017-05-11:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-05-11:
Hi
You are right, HSE value is defined in hal_conf file, which is shared in the family device directory...

You can have a look on the F7 hal conf file for ex:

#if defined(TARGET_DISCO_F746NG) || defined(TARGET_DISCO_F769NI)
  #define HSE_VALUE    25000000U /*!< Value of the External oscillator in Hz */
#else
  #define HSE_VALUE    8000000U /*!< Default value of the External oscillator in Hz */
#endif

closed 2017-05-13 by YarivCol