ARMmbed/mbed-os #4326: STM32F4xx:  Using AnalogIn() for internal VBAT and TEMP measurements

RobSenix, 2017-05-15:
analogin_api.c uses a strange conversion from 12 to 16 bit values in the read_u16() function:

    // 12-bit to 16-bit conversion
    value = ((value << 4) & (uint16_t)0xFFF0) | ((value >> 8) & (uint16_t)0x000F);
 
This produces a left shift of 4 and then copies bits 8-11 into the lowest 4 bits.  This makes the lowest 4 bits incorrect.  A better solution would be to take the sum of 16 readings, this would make the lowest 4 bits more meaningful.

Also seeing problems taking multiple readings from the internal VBAT and TEMP sensors on a MultiTech mDot device.  The first reading after reset is correct, then all others are not.  I have been able to fix the problem by stopping the ADC manually before every read:

            AnalogIn vBatIn(ADC_VBAT);

            // Disable the ADC before every reading to get valid values
            ADC1->CR2 &= ~ADC_CR2_ADON;   
            vBat = vBatIn.read_u16();

I've had to directly access the hardware because the AnalogIn class doesn't expose any way to modify the CR2 register.  It also doesn't implement any functionality in the class destructor.  This causes the ADC to always remain active and uses additional power.

Using mbed-os 5.4.2 with libmDot-mbed5 2.0.17-1.

0xc0170, 2017-05-16:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-05-19:
Hi,
Can you please split this issue in two in order to facilitate the tracking ?

The first part concerning the 12 to 16-bit conversion has been discussed a long time ago here #1767 and it has been agreed that this conversion model is ok. 

Thanks.

LMESTM, 2017-07-03:
@RobSenix 
Hi - thanks for reporting the issue here. I could reproduce and below is a patch for solving your issue
```
$ git diff targets/TARGET_STM/TARGET_STM32F4/analogin_api.c
diff --git a/targets/TARGET_STM/TARGET_STM32F4/analogin_api.c b/targets/TARGET_STM/TARGET_STM32F4/analogin_api.c
index 7947654..97cfbd4 100644
--- a/targets/TARGET_STM/TARGET_STM32F4/analogin_api.c
+++ b/targets/TARGET_STM/TARGET_STM32F4/analogin_api.c
@@ -128,7 +128,7 @@ static inline uint16_t adc_read(analogin_t *obj)

     // Configure ADC channel
     sConfig.Rank         = 1;
-    sConfig.SamplingTime = ADC_SAMPLETIME_15CYCLES;
+    sConfig.SamplingTime = ADC_SAMPLETIME_480CYCLES;
     sConfig.Offset       = 0;
```

I will work on the corresponding Pull Request shortly to fix the issue. I will probably address more than F4 family as similar issue has been solved recently as well on L0 family.

LMESTM, 2017-07-03:
Note: I have not reproduced the problem for TEMP measurement, I could only see it on VABT, so that #4691 does only increase sampling time for VBAT. Let me know if I missed something

LMESTM, 2017-07-19:
Considering that #4691 has been merged, I think this issue can be closed.
Let me know if this is fine with you.
ST_TO_BE_CLOSED

closed 2017-07-20 by 0xc0170