ARMmbed/mbed-os #4327: Hello,

marineDE, 2017-05-16:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

marineDE, 2017-05-16:
I cannot compile, from mbed compiler, a program using EthernetInterface.h from mbed-os (version 5.4.4.) because I get this error:
"Cannot open source input file "EthernetInterface.h": No such file or directory "#include "EthernetInterface.h""
 I checked the EthernetInterface is well in the mbed-os library and I can also include another library from there but not EthernetInterface . I tried to use the external EthernetInterface designed for LPC boards but of course a ton of other conflicts pop up.
Also, if I switch the platform from LPC4088 to K64F, it works. So what is wrong with LPC4088?

0xc0170, 2017-05-16:
@marineDE Please dont ignore the issue template , fill the details there , or just remove it completely. It would help :-)

To the issue, that target did not support lwip, it was added just recently, look at the commit  7c33ba8691683fb996abfc514b. Therefore the build system does not go into lwip featur

marineDE, 2017-05-16:
Yes sorry for the template..
But so what do I need to do if I want to have ethernet with LPC4088? Since I cannot use the EthernetInterface library (designed for LPC4088) since the mbed-os library overwrites on it? I do not understand why the target LPC4088 does not support LWIP since lwip is already inside the independant  EhernetInterface library (designed for LPC4088).

0xc0170, 2017-05-16:
It should be part of the release next week. You can locally or in the online IDE import latest mbed OS sources (development version) to test this feature in the meantime.

Please close this issue if this can be resolved

marineDE, 2017-05-18:
Thank you for your answer but is it really due to the LWIP library ? You said that target did not support lwip, but in the last release of mbed-os (5.4.5) I can see in the "fixes and changes":
[...] 4190: LPC4088: Enable LWIP feature [...]
 (https://docs.mbed.com/docs/mbed-os-release-notes/en/latest/5_4/release_note/)
With this latest version, I get 24 errors when I try to compile mbed-os with LPC4088 and all of these errors are due to "Undefined symbol" inside the mbedtls library... Do you think these errors will be fixed in the next release? Thank you in advance for your support!

0xc0170, 2017-05-18:
@marineDE Can you provide detailed issue report please? What errors, what toolchain? anything that we might help us to reproduce this issue.

btw, I run ``mbed test -m LPC4088 -t GCC_ARM -n features-feature_lwip-tests-mbedmicro-net-connectivity --compile`` , it builds as it should. This was tested locally on the latest master branch.

We can reopen this once we are able to reproduce the issue

closed 2017-05-18 by 0xc0170

marineDE, 2017-05-19:
I imported in the mbed compiler the HelloMQTT program of Vergil Cola, I updated the mbed-os library to the latest one and I removed the easy-connect library (I just kept and added inside the main file, the easy_connect function for Ethernet). If I try to compile for LPC4088, I get 24 similar errors:

[image]

So I tried to add the following macro in the "mbed_app.json" file:
"macros": ["MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES","MBEDTLS_TEST_NULL_ENTROPY"]
With this it compiles but I cannot get any output, even if I have only a printf in the main function...

0xc0170, 2017-05-19:
Ok so this is a problem of mbedtls and lpc4088, please open a new issue to request mbedtls support for this device (thats the one that is failing for MTQTTT).

marineDE, 2017-05-19:
Finally even after adding the macros:
"macros": ["MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES","MBEDTLS_TEST_NULL_ENTROPY"]
I got the same 24 errors as well... (I fixed the problem for which I could not get any outputs, it was due to the arguments of the main function). Any help regarding these errors ("Undefined symbols [...]" ) would be appreciated...

marineDE, 2017-05-19:
OK Thank you I will open a new issue!