ARMmbed/mbed-os #4330: No rtos support

YarivCol, 2017-05-16:
### Description
- Type: Question
- Related issue: `#4316`

## Question
will mbed support no rtos configuration ?
Because i saw in issue [4316](https://github.com/ARMmbed/mbed-os/pull/4316) that the cmain.s was removed, this file was for the use of no rtos startup.

0xc0170, 2017-05-17:
Can you be more specific? What I am missing is what mbed do you mean?

This might answer your question:

mbed 2 stays as it is, no rtos by default. Therefore there will be own startup. The new startup for mbed OS will be part of rtos, and contains rtos specific startup.

> Because i saw in issue 4316 that the cmain.s was removed, this file was for the use of no rtos startup.

I could not set a breakpoint when I was testing it (I had a local branch reverting this removal), I had an interested to check  the arg function there (``__iar_argc_argv``), as I found out there was no invocation of that function. I used IAR v7.80. What version of IAR are you using? Is cmain.S used and that ``__iar_argc_argv`` invocation is actually happening ?

YarivCol, 2017-05-17:
actually the __cmain function preforms the boot process,  the implementation resides in cmain.s file so no boot process other than iar default boot (no call to sdk_init, etc..). 
the main will be called at the end but mbed will not be initialized.

0xc0170, 2017-05-17:
Thats correct.

I will send a PR, with reverting the removal, tested locally.

0xc0170, 2017-05-17:
PR sent with a fix, please review

PR #4334 (2017-05-22): Revert partially "Remove Cortex A support from CMSIS/RTOS"
This reverts commit 5f294451030ee1f7982afbb41c2badb7b986d822 - but only partially.

cmain.S for IAR is still required to keep mbed SDK boot process as it was. This
startup file for IAR is active for non-rtos projects.

Fixes #4330

Thanks to @YarivCol for reporting this. Tested locally with blinky mbed OS and mbed-dev gpio example (K64F board) - thus should be compatible with all new changes (new mbed boot that is coming to mbed OS).
commit e0dded3b (2017-05-22): Revert partially "Remove Cortex A support from CMSIS/RTOS"

This reverts commit 5f294451030ee1f7982afbb41c2badb7b986d822 - but only partially.

cmain.S for IAR is still required to keep mbed SDK boot process as it was. This
startup file for IAR is active for non-rtos projects.

Fixes #4330
commit 8c7df97b (2017-05-22): Revert partially "Remove Cortex A support from CMSIS/RTOS"

This reverts commit 5f294451030ee1f7982afbb41c2badb7b986d822 - but only partially.

cmain.S for IAR is still required to keep mbed SDK boot process as it was. This
startup file for IAR is active for non-rtos projects.

Fixes #4330
closed 2017-05-22 by 0xc0170