ARMmbed/mbed-os #4339: When is planned the update of STM32cubeL4 driver ?

pmancele, 2017-05-18:
### Description
- Type: Question
---------------------------------------------------------------
**Target** : STM32L4xx
----------------------------------------------------------------
## Question:
We are working on new STM32L496ZG and NUCLEO_STM32L496ZG. Previously we were working on STM32L476ZG that we successfully ported on mbed-os.
But we can't port the STM32L496ZG as it needs a more recent version of the ST HAL driver "STM32cubeL4". I tried to udpate it manually from 1.5 to 1.7 and it worked for the 496 but i lost compatibility with other L4 targets..
So I would like to know if ST is planning to update the HAL driver for the L4 family in mbed-os ?
Thank you !

jeromecoutant, 2017-05-18:
Hi

OK, I can start to update the HAL driver for L4 family now.

About NUCLEO L496 new port, I am sorry, it is not currently planned by ST...

@bcostm

bcostm, 2017-05-18:
Yes I confirm that the support of the NUCLEO_L496ZG board is not planned to be done by ST.

jeromecoutant, 2017-05-19:
Please check #4351 

About NUCLEO_L496ZG, I can provide you the PeripheralPins.c
You have to comment several lines in this fiel in order to remove not available pins (USB, debug uart, ... pins)
[PeripheralPins.zip](https://github.com/ARMmbed/mbed-os/files/1013820/PeripheralPins.zip)

ohagendorf, 2017-05-19:
Hello @jeromecoutant , @bcostm 

are there any plans to extend the PeripheralPins.c in the way I sugested in #3181
Somebody of you mentioned that there is a tool which generates the PeripheralPins.c automatically, when I understood correctly.

bcostm, 2017-05-19:
Hi @ohagendorf 

> are there any plans to extend the PeripheralPins.c in the way I sugested in #3181

Yes we have started to do it on new targets like the DISCO_L072CZ and DISCO_L475VG but nothing planned for now on actual targets.

pmancele, 2017-05-19:
@jeromecoutant Thank you for the PeripheralPin file and the update !
I will try this asap.

ohagendorf, 2017-05-19:
@bcostm , thanks for the info. I can do this manually for the current targets, if the daily work permits it.

bcostm, 2017-06-06:
Hi @pmancele 
The Cube L4 has been updated in #4351. Are you happy to close this issue now ? Thanks.

pmancele, 2017-06-06:
Hi @bcostm Yes, thank you for the update ! 
Thanks to it, we managed to port our board base on STM32L476ZG without loosing compatibility with other boards.

Regards

closed 2017-06-06 by pmancele