ARMmbed/mbed-os #4341: HAL_LOCK in ST Devices

daniel-cesarini-tridonic-com, 2017-05-18:
### Description
- Type: Bug
- Priority: Major

HAL_LOCK mechanism in ST HAL seems not ready for RTOS context.

---------------------------------------------------------------
## Bug

**Target**
STMF32F0 and F4

**Toolchain:**
All

**mbed-os sha:**
4eea8fa8633751fc91095e24eeb154ac3df3afd9

**Actual behavior**
The HAL_LOCK mechanism (https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_def.h#L105) is not set to be used in the RTOS context. 

**Observed issue**
When using the CRC hardware for checking data coming in from an I2C interrupt and from a scheduled thread we observed not proper behavior on the handling of HAL_LOCK.
We are working on a workaround, to schedule the CRC check outside of the interrupt context, but are still unsure whether this solution then will completely overcome the limitations of the current implementation of the HAL_LOCK mechanism.

-----------------------------------------------------------------

## Question

**How to?**
How to use the ST HAL in the context of mbed-os considering that it is an RTOS? And as such it could access the HAL from both interrupts and thread context.
Related to this I found https://community.st.com/thread/13989, where also it is reported that the HAL_LOCK mechanism is not completely reliable in the context of RTOS systems.

@bcostm
-------------------------
FYI: @markus-becker-tridonic-com @ivan-adzic-tridonic-com @MarceloSalazar @karsev

0xc0170, 2017-05-18:
cc @bcostm @adustm @LMESTM @jeromecoutant

LMESTM, 2017-05-18:
Hello - as mentioned in the community post, a new LOCK and UNLOCK process under preparation. We'll keep you posted and get updates to mbed when this is available.

bcostm, 2017-05-18:
ST_INTERNAL_REF 16296

daniel-cesarini-tridonic-com, 2017-05-19:
@LMESTM @bcostm by when do you expect that the new LOCK and UNLOCK mechanisms should be ready for being used? It looks to me like that community post is already 1 year old. 

Other questions: 
* Do you expect that the HAL might have other issues in the current status of implementation when being used in the mbed-os (RTOS) context? 
* Would it be possible that accessing the HAL from different threads / interrupts might have bad consequences (e.g. inconsistent state, faults, ...)?

FYI: @markus-becker-tridonic-com @ivan-adzic-tridonic-com @MarceloSalazar

0xc0170, 2017-05-19:
> Other questions:

    Do you expect that the HAL might have other issues in the current status of implementation when being used in the mbed-os (RTOS) context?
    Would it be possible that accessing the HAL from different threads / interrupts might have bad consequences (e.g. inconsistent state, faults, ...)?

Regarding the above even, how is i2c used  ? Is it within I2C class?

Because it should be thread safe. HAL is not, but the upper layer should be. In case it's not, please provide an example with description

daniel-cesarini-tridonic-com, 2017-05-19:
Actually we implemented also a CRC Class, but did not consider that the HAL is not thread safe. 

Furthermore I am currently checking the I2C Class for possible issues (e.g., what would happen if an I2C operation would be called from within interrupt context?).

0xc0170, 2018-06-15:
@ARMmbed/team-st-mcd  Is this still relevant or the update happened but this stayed opened?

LMESTM, 2018-06-15:
@daniel-cesarini-tridonic-com  @0xc0170 
JHAL_LOCK update has not been delivered yet. In the context of mbed-os, it is expected that drivers (I2C or CRC driver ensure the thread safety protection which is done with the driers" lock.

0xc0170, 2018-06-15:
Thanks, this can be then resolved with provided answers from myself above and @LMESTM

@daniel-cesarini-tridonic-com If there is anything we have missed, please reopen the issue.

closed 2018-06-15 by 0xc0170

daniel-cesarini-tridonic-com, 2018-06-18:
@LMESTM @0xc0170 : I did not completely capture the new situation:
what is expected to be done by ST? By ARM? By the final user?

@LMESTM reported that finally the HAL_LOCK is not delivered yet. 
Is thread safety to be implemented with Semaphores / Mutexes? If so, will ARM take care of this? Or has it already been done?

Thanks!

LMESTM, 2018-06-18:
@daniel-cesarini-tridonic-com 
The HAL_LOCK will be updated in ST drivers (CUBE)  so that it will be thread safe as well in the future. 
Even if not done yet, this is not blocking in mbed-os context where the mbed-os drivers are taking care of ensuring thread safety,  which is done like in I2C driver here: 
https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2C.cpp#L45
So if you develop you own mbed CRC driver, you should follow the same guidelines and implement a lock/unlock in this driver.