ARMmbed/mbed-os #4347: Parity is not enabled

matsujirushi, 2017-05-19:
### Description
- Type: Bug
- Related issue: https://developer.mbed.org/questions/78052/Parity-is-not-enabled/
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-L476RG

**Toolchain:**
ARM (online compiler)

**Expected behavior**
Parity should be enabled when it is specified by Serial::format() function.

**Actual behavior**
Parity is not enabled.

**Steps to reproduce**

Example code:
#include "mbed.h"
 
Serial uart(PA_9, PA_10);
 
int main()
{
    uart.baud(9600);
    uart.format(8, Serial::Even, 1);
 
    for (;;)
    {
        uart.puts("ABC");
        wait_ms(200);
    }
}

Result:
https://developer.mbed.org/media/uploads/matsujirushi/20170517a.png

matsujirushi, 2017-05-19:
Fix example:
mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart.c
```c++
  /*-------------------------- USART CR1 Configuration -----------------------*/
  /* Clear M, PCE, PS, TE, RE and OVER8 bits and configure
   *  the UART Word Length, Parity, Mode and oversampling:
   *  set the M bits according to huart->Init.WordLength value
   *  set PCE and PS bits according to huart->Init.Parity value
   *  set TE and RE bits according to huart->Init.Mode value
   *  set OVER8 bit according to huart->Init.OverSampling value */
  if (huart->Init.Parity == 0)
  {
	  tmpreg = (uint32_t)huart->Init.WordLength | huart->Init.Parity | huart->Init.Mode | huart->Init.OverSampling;
  }
  else
  {
	  tmpreg = (uint32_t)huart->Init.WordLength | huart->Init.Parity | huart->Init.Mode | huart->Init.OverSampling | USART_CR1_M0;
  }
  MODIFY_REG(huart->Instance->CR1, UART_CR1_FIELDS, tmpreg);
```

LMESTM, 2017-05-19:
@matsujirushi  the management of parity bits is being corrected in #4216
can you check the proposed fix in #4216  ?

matsujirushi, 2017-05-20:
I tested #4216. It's fine.

9600-8-N-1
[image]

9600-8-E-1
[image]

9600-8-O-1
[image]

LMESTM, 2017-05-22:
@matsujirushi thank you for your feedback. so I propose to close this ticket , either now or when #4216 is merged

matsujirushi, 2017-05-22:
OK. Thank you.

closed 2017-05-22 by matsujirushi