ARMmbed/mbed-os #4349: nrf51 targets - exports contain duplicated files

0xc0170, 2017-05-19:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
possibly any nrf51 targets?

**Toolchain:**
any, tested with IAR

**meed-os sha:**
5d0ce3c

**Actual behavior**

I can see duplicated files, this has something to do with unified introduction, here is IAR report:

```
Building configuration: mbed-os-example-blinky - mbed-os-example-blinky 
Updating build tree... 
Build error: Multiple tools write to the same file. 
  File: C:\Code\examples\mbed-os-example-blinky\BUILD\Obj\analogin_api.o 
  Tool 1: C/C++ Compiler, when processing C:\Code\examples\mbed-os-example-blinky\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5\analogin_api.c 
  Tool 2: C/C++ Compiler, when processing C:\Code\examples\mbed-os-example-blinky\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5\ 
TARGET_MCU_NRF51822_UNIFIED\analogin_api.c 
 
Build aborted. 
```

@nvlsianpu @pan-

nvlsianpu, 2017-05-19:
@0xc017 I thing a rename one of those file to other name name will resolve the issue. What do you think?

0xc0170, 2017-05-19:
This is the merge that breaks it : https://github.com/ARMmbed/mbed-os/pull/4245

I wonder why nrf51 has these in targets.json file ``        "extra_labels": ["NORDIC", "MCU_NRF51", "MCU_NRF51822_UNIFIED", "NRF5", "SDK11"] `` . NRF and UNIFIED are causing this.

What's the intention of NRF5/ and NRF5/MCU_NRF51822_UNIFIED (why there are 2 implementations for HAL in those 2) ?

0xc0170, 2017-05-19:
> @0xc017 I thing a rename one of those file to other name name will resolve the issue. What do you think?

If I understand correctly, there are two implementations of HAL. One of these is protected inside via macro so always only one is compiled? This then breaks in IAR. Correct?

Rename might work but it is confusing for a user, like you can see for me also, 2 labels, 2 implementations of HAL but one only gets to the compiliation and further. Is there any other way we can fix it , using labels for instance ?

0xc0170, 2017-05-19:
Note: we shall run export build always, this was not caught because it was not run, it would definitely fail to build.

0xc0170, 2017-05-19:
> Note: we shall run export build always, this was not caught because it was not run, it would definitely fail to build.

@studavekar Can we do that? ``/morph test`` would also include exporters ?

nvlsianpu, 2017-05-19:
> If I understand correctly, there are two implementations of HAL. One of these is protected inside via macro so always only one is compiled? This then breaks in IAR. Correct?

Yes I is. As I see IAR generate all objects files to one directory :-(

From the point of view of nRF5x SoCs: nRF52832 and nRF52840 both have the same ADC peripheral, hence the same low level driver and have the same hal driver api implementation. nRF51xxx just opposite. Therefore there is no sense to put implementations together in one file.

So 1st proposition is to rename the file, 2nd is to add another extra_label for differ one implementation from another. Do you prefer the second one? (I suppose you do thanks to above ceinversation)

0xc0170, 2017-05-19:
> So 1st proposition is to rename the file, 2nd is to add another extra_label for differ one implementation from another. @0xc0170 Do you prefer the second one? (I suppose you do thanks to above ceinversation)

would more intuitive in the IDE workspace - thus I would go with it.

pan-, 2017-05-22:
@nvlsianpu Any update on this issue ?

nvlsianpu, 2017-05-23:
I have this in my task queue.

nvlsianpu, 2017-05-23:
I see that resolution is pending.

PR #4366 (2017-05-24): Move duplicated analogin and pwmout to NRF52 directory
# Description
Resolves #4349

Both `analogin_api.c` and `pwmapi.c` would fail for any `MCU_NRF51822_UNIFIED` 
target, as the files were duplicated there, when they seem to be intended for
any `MCU_NRF52` target. This PR moves the files from the root to the latter
target.

# Status
**Ready**

# Todos
- [ ] /morph test
- [ ] /morph export-build
closed 2017-05-24 by 0xc0170