ARMmbed/mbed-os #4350: Issues with MbedTLS and LPC4088

marineDE, 2017-05-19:
I try to compile mbedTLS with LPC4088 but I got systematically these errors:
[image]

Actually, I imported in the mbed compiler the HelloMQTT program of Vergil Cola, I updated the mbed-os library to the latest one and I removed the easy-connect library (I just kept and added inside the main file, the easy_connect function for Ethernet). Also, I added in the mbed_app.json file this macro:
"macros": ["MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES","MBEDTLS_TEST_NULL_ENTROPY"]

But I still cannot compile... Is there any issue for mbedTLS with LPC4088?
Thank you in advance for any help!

0xc0170, 2017-05-19:
cc @sbutcher-arm

0xc0170, 2018-06-15:
@marineDE I believe you resolved you issue?  The issue is also missing vital info - what mbed OS version, small test program that we can reproduce the failure.

I haven't experienced any problems with latest Mbed OS and compiling some mbedtls tests.

Closing due to inactivity and lack of details. Please reopen if this issue is still relevant.

closed 2018-06-15 by 0xc0170