ARMmbed/mbed-os #4354: MOTE_L152RC Target:  DigitalOut

snelson-senet, 2017-05-19:
### Description
- Type: Bug 
---------------------------------------------------------------
## Bug
DigitalOut interface is broken.  In my case, controlling LEDs stops working.  I do not see this failure when I switch to an mbed-os  revision using mbed library version <= 134

**Sample code:**
DigitalOut RedLed(LED1);

// LED is active Low
RedLed = 0 ;    // This works (LED  ON)
RedLed = 1;     //  This works (LED OFF)
RedLed = 0;     //  Now broken, LED remains OFF forever

**Target**
MOTE_L152RC

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4-2016q3

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
8921974 Merge pull request #4278 from ARMmbed/release-candidate

theotherjimmy, 2017-05-19:
Hi @snelson-senet I changed the title to match the description.

theotherjimmy, 2017-05-19:
@0xc0170 It looks like this is an STM32, and only supports mbed OS 2.

snelson-senet, 2017-05-24:
I created pull request https://github.com/ARMmbed/mbed-os/pull/4378 to show how I fixed this issue.
Feel free to use if feasible.

cmonr, 2017-12-14:
Closed since PR has long since been merged.

closed 2017-12-14 by cmonr