ARMmbed/mbed-os #4355: changing program start address

mohammadmh, 2017-05-20:
Hi guys,
I've exported mbed_blinky project for keil uvision 5 and i want to develop a program for another board, so I need to change the start address of program. I change the start in IROM1 in options for target window but it doesn't work. It must be defined in somewhere but I can't find it. Can anyone help?
*My target is LPC1768

theotherjimmy, 2017-05-22:
If you want to develop your program for another board, I recommend that you re-export it for that board. If the board is not in mbed and you're not concerned with ever passing this program to someone else, you could change the scatter file to have the correct address. The scatter file is here (relative to mbed-os) `./targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_ARM_STD/LPC1768.sct`

theotherjimmy, 2017-05-22:
Let me know if that helps, @mohammadmh

cmonr, 2017-12-14:
Closing since question appears to be resolved.

closed 2017-12-14 by cmonr