ARMmbed/mbed-os #4361: [Nucleo-F070RB] Wrong divisor when use internal clock

mickeyln, 2017-05-22:
### Description
- Type: Bug
- Related issue: 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
ST32M|Nucleo-F070RB

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
Remove external clock will change to use internal clock automatically. And it clock will be 48Mhz.

**Actual behavior**
The clock become to 96Mhz or no work.

**suggest **
It seems to me that there is an error in the mbed library for F070RB.

File:
targets/cmsis/TARGET_STM/TARGET_STM32F0/TARGET_NUCLEO_F070RB/ system_stm32f0xx.c 
Function:
SetSysClock_PLL_HSI (line431)

change: 
**RCC_OscInitStruct.PLL.PREDIV              = RCC_PREDIV_DIV1;**

to:
**RCC_OscInitStruct.PLL.PREDIV              = RCC_PREDIV_DIV2;**

0xc0170, 2017-05-22:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-05-23:
Hi,
I have reproduced the issue with HSI. The system clock does not start. The error is well on the line you mentionned. Do you mind sending a pull-request to fix it ? Thanks.

PR #4390 (2017-06-26): Nucleo-F070RB It doesn't work when use internal clock
## Description
Nucleo-F070RB have a wrong divisor when it initial the clock.

## Status
**READY**

## Migrations
If this PR changes any APIs or behaviors, give a short description of what *API users* should do when this PR is merged.

**NO**

## Steps to test or reproduce
Run display message to UART

Fixes #4361
closed 2017-06-26 by theotherjimmy