ARMmbed/mbed-os #4372: LWIP K64F ipv6 ethernet does not work with arm compiler optimizations

mikaleppanen, 2017-05-23:
### Description
- Bug
- Minor

## Bug

**Target**
K64F

**Toolchain:**
ARM

**Toolchain version:**
All 5.06 versions

**Expected behavior**
LWIP K64F ipv6 ethernet works.

**Actual behavior**
ARM compiler compiles K64F ethernet driver code incorrectly when both -O3, -Otime optimizations are enabled. Error happens in fsl_enet.c modules function ENET_AddMulticastGroup() crc calculation loop.
Compiler issue has been raised and will be fixed to next compiler release.

**Steps to reproduce**
Test LWIP K64F ipv6 ethernet. IPv6 address cannot be acquired.

0xc0170, 2017-05-25:
@mikaleppanen Compiler issue ? can you be more specific?  I assume this is to inform users, and might be worth adding to the known issues?

cc @adbridge

0xc0170, 2017-05-25:
@mmahadevan108 to be aware

JammuKekkonen, 2017-09-27:
@JanneKiiskila @teetak01

teetak01, 2017-09-27:
@mikaleppanen is this still open? On what compilers would this work?

mikaleppanen, 2017-09-27:
It would work on any other combination than described above i.e. arm compiler -O3, -Otime. There has not been update to arm compiler 5 that would solve this. Most probably arm compiler 6 does not have this issue.

JanneKiiskila, 2017-10-05:
ARM CC v5.06 update 6 has this fix included to it, we recommend updating your compiler.

0xc0170, 2017-10-05:
@mikaleppanen If you are happy with the proposal, please close the issue.

@JanneKiiskila we should make sure our CI have this installed?

JanneKiiskila, 2017-10-05:
Our CI has been taken care of already by @miklis.

closed 2017-10-06 by mikaleppanen