ARMmbed/mbed-os #4374: Maximum COAP message resending buffer size limited to 255 bytes.

Apocalyt, 2017-05-23:
### Description

- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

In the existing solution it is impossible for the maximum size of the COAP retransmission buffer to be bigger than 255 bytes. If you try to define YOTTA_CFG_COAP_RESENDING_QUEUE_SIZE_BYTES to be larger than 255 in application the value gets truncated to a uint8_t. This is probably because the sn_coap_resending_queue_bytes -variable is defined as uint8_t in:
mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/include/sn_coap_protocol_internal.h 

The maximum size of 255 bytes for the whole retransmission buffer can be too small for many applications especially when using COAP over WIFI or cellular network where multiple messages may require retransmission simultaneously. Even a single packet with a big payload (~512 bytes) can fill the whole buffer. The RAM available on many microcontrollers can easily handle a bigger buffer, so it should be possible to set the maximum size of the retransmission buffer to be bigger atleast using yotta config.

**Suggested enhancement**
Change the type of the sn_coap_resending_queue_bytes to be bigger to allow a bigger retransmission buffer if the programmer using the library so chooses. uint16_t or maybe even uint32_t should work better. The related unit test may also need to be changed.

**Pros**
- Allows bigger coap retransmission buffer size if the user so chooses
- Makes coap retransmission feature relevant in applications which require big amounts of data to be sent over unreliable network.

**Cons**
- Few (~1-3) extra bytes of ram usage when using COAP to store the bigger variable

0xc0170, 2017-05-25:
cc @SeppoTakalo @JanneKiiskila

Apocalyt, 2017-06-13:
Hello, what is the situation regarding this change request?
Has someone started to implement it yet?
I can try to provide some additional info if needed.

anttiylitokola, 2017-08-16:
Hi,

Thanks for you report. We haven't yet started to implementing this change but it will be part of upcoming releases.

Thanks,
Antti

anttiylitokola, 2017-09-13:
This fix is now part of mbed-os 5.5.6 release.

0xc0170, 2018-06-15:
> This fix is now part of mbed-os 5.5.6 release.

Resolved

closed 2018-06-15 by 0xc0170