ARMmbed/mbed-os #4380: Update vscode exporter for the new intellisense engine

janjongboom, 2017-05-24:
https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/LanguageServer/FAQ.md#what-is-the-difference-between-includepath-and-browsepath-in-c_cpp_propertiesjson

The intellisense engine has changed and now requires us to set a number of extra parameters and macros. For now we'll be on the old engine, but we should update the exporter.

janjongboom, 2017-12-15:
@cmonr Why is this being closed?!

cmonr, 2017-12-15:
@janjongboom I was under the impression that https://github.com/ARMmbed/mbed-os/pull/4381 fixed this issue. Is this not the case?

janjongboom, 2017-12-18:
@cmonr No, it's a workaround that forces vscode to use the *old* intellisense engine. We need to start using the new one though.

cmonr, 2017-12-19:
@theotherjimmy

theotherjimmy, 2017-12-19:
@janjongboom @cmonr I have no plans to contribute this enhancement at this time.

janjongboom, 2017-12-20:
@theotherjimmy @cmonr You can assign this to me.

closed 2019-10-31 by ciarmcom