ARMmbed/mbed-os #4384: Nucleo-f446re 4-6 second Timeout

mr63, 2017-05-24:
- Type: Bug 
- Priority: Blocker
---------------------------------------------------------------
## Bug

**Target**
Nucleo-f446re 

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
MBED -- Revision 133:99b5ddf

I noticed today after creating a simple Hello World application that any programs that I compile with mbed revisions after 132:9baf128 seem to take 4-6 seconds to start.  I noticed via google https://github.com/ARMmbed/mbed-os/issues/1096
that this was a problem in very early versions of the nucleo versions, but has been corrected.

0xc0170, 2017-05-25:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-06-01:
Hi
Are you using HSE (default configuration) or HSI ?
Regards,

jeromecoutant, 2017-06-01:
@mr63 , please have a look on #4422

jeromecoutant, 2017-06-02:
My first question is still valid :
Are you using HSE (default configuration) or HSI ?
Do you use default NUCLEO board or you changed some jump ?
Thx

mr63, 2017-06-02:
I am using the default Configuration.

Mark

mr63, 2017-06-02:
Like said in my post.  The previous version of the library works correctly.

jeromecoutant, 2017-06-07:
Hi
I don't understand...
Default board configuration is using HSE: after reset, you start immediatly with HSE, no delay is observed.
If your HSE is not working, SetSysClock procedure is waiting for the HSE timeout value, and then start with HSI/MSI: this timeout value was 5s which was a wrong value, #4422 is correcting this.

jeromecoutant, 2017-06-07:
Hi
This is also coherent with your description:
With MBED revision 132, HSE timeout value was 200 ms,
with MBED revision 133, HSE timeout value has been set to 5s

jeromecoutant, 2017-06-27:
#4422 is now merged, and will be integrated in 5.5.2

 ST_TO_BE_CLOSED

closed 2017-06-27 by 0xc0170