ARMmbed/mbed-os #4391: should we add interface to set SOF_BROADCAST?

kyliuxing, 2017-05-25:
@kjbracey-arm 
we have enabled IP_SOF_BROADCAST and IP_SOF_BROADCAST_RECV, it will check SOF_BROADCAST when we send broadcast packet, but we didn't add this opt in mbed_lwip_setsockopt().
Should we add it? or we won't send broadcast packet.

kjbracey-arm, 2017-05-29:
Why have you enabled those? 

As the port stands, the NSAPI C++ layers do not provide SO_BROADCAST, so applications cannot, and do not have to, set SO_BROADCAST to send (or receive) broadcasts. NSAPI requires those flags to be off.

If you were to enable them, to maintain backwards application compatibility NSAPI would have to turn SO_BROADCAST on by default, which would basically defeat the purpose of the flags (a safety lock to prevent unintended broadcasts).

If this is for your own private applications, so you're not concerned with backwards compatibility, then you could add an SO_BROADCAST socket option that doesn't default on to NSAPI in your fork. But I don't currently expect the mainline code to pick that up.

Note that IPv6 has no equivalent of SO_BROADCAST, and it doesn't control IPv4 multicast either. So it feels like a legacy option. And that lwIP RECV flag is rather unusual - no conventional network stack prevents broadcast reception that way.

kyliuxing, 2017-05-31:
@kjbracey-arm 
Thanks for your help, i will find a appropriate way to resolve my problem.

closed 2017-05-31 by kyliuxing