ARMmbed/mbed-os #4399: Error while compiling mbed_retarget.cpp for KL25Z with IAR 8.11

1807374010, 2017-05-28:
[Error] mbed_retarget.cpp@911,47: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@911,15: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@912,30: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@912,15: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@913,30: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@913,15: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@926,47: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@926,15: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@927,30: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@927,15: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@928,30: [Pe393]: pointer to incomplete class type is not allowed
[Error] mbed_retarget.cpp@928,0: [Pe393]: pointer to incomplete class type is not allowed

          _file->_Mode = (unsigned short)(_file->_Mode & ~ 0x1000);/* Unset read mode */
               ^
"F:\mbed\platform\mbed_retarget.cpp",911  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Mode = (unsigned short)(_file->_Mode & ~ 0x1000);/* Unset read mode */
                                               ^
"F:\mbed\platform\mbed_retarget.cpp",911  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Rend = _file->_Wend;
               ^
"F:\mbed\platform\mbed_retarget.cpp",912  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Rend = _file->_Wend;
                              ^
"F:\mbed\platform\mbed_retarget.cpp",912  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Next = _file->_Wend;
               ^
"F:\mbed\platform\mbed_retarget.cpp",913  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Next = _file->_Wend;
                              ^
"F:\mbed\platform\mbed_retarget.cpp",913  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Mode = (unsigned short)(_file->_Mode & ~ 0x1000);/* Unset read mode */
               ^
"F:\mbed\platform\mbed_retarget.cpp",926  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Mode = (unsigned short)(_file->_Mode & ~ 0x1000);/* Unset read mode */
                                               ^
"F:\mbed\platform\mbed_retarget.cpp",926  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Rend = _file->_Wend;
               ^
"F:\mbed\platform\mbed_retarget.cpp",927  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Rend = _file->_Wend;
                              ^
"F:\mbed\platform\mbed_retarget.cpp",927  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Next = _file->_Wend;
               ^
"F:\mbed\platform\mbed_retarget.cpp",928  Error[Pe393]: pointer to incomplete class type is not allowed

          _file->_Next = _file->_Wend;
                              ^
"F:\mbed\platform\mbed_retarget.cpp",928  Error[Pe393]: pointer to incomplete class type is not allowed

Completed in: (57.92)s

Build failures:
  * IAR::KL25Z

1807374010, 2017-05-28:
I think maybe the mbed-os doesn't support the newest IAR EWARM.In the newest IAR the freescale has been replaced by NXP.

0xc0170, 2017-05-30:
> I think maybe the mbed-os doesn't support the newest IAR EWARM.In the newest IAR the freescale has been replaced by NXP.

That is correct. the IAR template files are targeting IAR 7.8, therefore it can happen that IAR 8.x might contain wrong configuration (as you noted NXP change, and possibly others). Please use 7.8 version

I'll close this as it is not supported.

closed 2017-05-30 by 0xc0170