ARMmbed/mbed-os #4404: STM L072CZ_LRWAN1 PC13 Pin missing, target not working in the Online IDE

helmut64, 2017-05-30:
The PinNames.h does not specify PC_13, however the PC_13 is available and being connected to the Lora radio DIO3 interrupt which is required. All other Lora Pins are defined.

The DISCO-L072CZ-LRWAN1 taget is selectable for the online IDE, however brings the message:
Error: Target "DISCO_L072CZ_LRWAN1" is not recognised

Once you have it ready I can test my supported SX1276GeneribLib for this board and provide further feedback.
Regards Helmut

0xc0170, 2017-05-30:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-05-30:
Hi Helmut,

Thanks for the info. For the PC_13 pin, it's a mistake we'll correct it ASAP.

Concerning the error message it is strange because on my side I don't have any problem. I can compile a project with this target without any error. Just to be sure, did you click on the button "Add to your mbed Compiler" and do you use the mbed rev 143 ?

helmut64, 2017-05-30:
Very good, the target is working in the online IDE (I missed to updated mbed fro this project). However it is not listed in the boards list of the mbed web site.

I updated my working Lora PingPong example which supports your board based on my SX1276Generic driver which received support for the Murata lora chip used in the LRWAN1.
http://developer.mbed.org/users/Helmut64/code/STM32L0_LoRa/

I will do further testing checking out the sleep and deelseep modes.

I close this bug because I have seen your checkin.

Regards Helmut

closed 2017-05-30 by helmut64

bcostm, 2017-05-31:
> However it is not listed in the boards list of the mbed web site.

Yes I know and ARM will add it by the end of the week.

And thanks again for your Lora PingPong example.

Just some remarks on this demo: the HSI clock configuration can be removed now in main.cpp because it is already done in system_l0xx.c file, and I think you can use now the macro TARGET_DISCO_L072CZ_LRWAN1.

helmut64, 2017-05-31:
Yes I will update the project after I completed some additional testing.

HarishkumarBringiramakannu, 2018-06-12:
Hi Helmut !
I am using B-L072Z-LRWAN1 device with online mbed compiler. But for some of my programs,I get error as ( Error: Target "DISCO_L072CZ_LRWAN1" is not recognized ). Do you know why ? Can you explain briefly to how to recover from his ?
Currently I am working with DS18B20 1-wire temperature sensor. If you have the code, you can suggest me, so I can try.

helmut64, 2018-06-13:
Dear HarishkumarBringiramakannu, this is not the right place for you question, you are welcome to contact me via an private message within the mbed portal. Please ensure that your mbed profile is completed with contact details because I don't discuss with anonymous people.