ARMmbed/mbed-os #4407: Is there a canonical sequence of operations that will always open a (possibly corrupt) FatFileSystem?

NeilMacMullen, 2017-05-30:
UBLOX_EVK_NINA_B1
GCC_ARM
MbedOS 5.4.5

I have a FatFileSystem mounted on a serial-flash.  I have been unable to find a reliable sequence of operations that will reliably open the filesystem when the flash is in an unknown state.  What I would _like_ to be able to do is...

1) If the flash contains a valid, uncorrupted filesystem image, open it and start using it.
or 2) If the flash is corrupt/unformatted, format the flash then open it and start using it.

I have tried variations on sequences such as
```
flsh = new SPIFBlockDevice(..flash driver...);
fs = new FATFileSystem("fs");
if(fs->mount(flsh))
{
   //unmount first?
    FATFileSystem::format(flsh);
    //remount??
}
```
including deleting and reconstructing the SPIBlockDevice and filesystem, but once I've made the first mount attempt it seems to be impossible to successfully format and then start using the filesystem.

The only reliable way I've found of doing this is....

```
//format has to be done before mount ? ....
if (NonVolatile.NeedsFormat)   
        FATFileSystem::format(flsh);
error = fs->mount(flsh);
if (error)  
        {NonVolatile.NeedsFormat=true; ResetBoard();}
//strangely, we seem to be able to mount corrupt file systems so try reading the root
//directory as a check...
DIR *dir = opendir("/fs/");
if (!dir)
      {NonVolatile.NeedsFormat=true; ResetBoard();}
//if we get this far, filesystem is good (we hope)
```

Is there some sequence of operations provided by the FFS API that allows this to be done more gracefully?

MarceloSalazar, 2017-05-30:
@geky this might be caused by the init/deinit on the block device

geky, 2017-06-14:
Hi @NeilMacMullen, sorry about the delay.

Your example _should_ be correct. However there was a bit of an issue around block device initialization that was unexpected. Current master requires block device init/deinit outside of the mount and format functions.

I put up a fix that should take care of this: https://github.com/ARMmbed/mbed-os/pull/4562. If your first example doesn't work with this patch let me know.

NeilMacMullen, 2017-06-15:
Thanks @geky - will give this a try in the next few days.

geky, 2018-03-14:
I believe this has been solved. Feel free to reopen if that's not the case

closed 2018-03-14 by geky