ARMmbed/mbed-os #4419: Cannot compile Target NUCLEO_F746ZG with mbed-cli

scamille, 2017-06-01:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F746ZG

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)

**mbed-cli version:** 
1.1.1

**meed-os sha:**
Both non-working:
5fff7e1 Merge pull request #4356 from ARMmbed/release-candidate
88c398b Merge pull request #4377 from theotherjimmy/disable-cortex-A

**Program content**
A single main.cpp file with int main() { return 0; }

**Expected behavior**
Compile & Linking without errors

**Actual behavior**
Linking fails:
```
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o): Conflicting CPU architectures 13/1
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o)
collect2: error: ld returned 1 exit status
[ERROR] /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o): Conflicting CPU architectures 13/1
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o)
collect2: error: ld returned 1 exit status
```

-vv
```
[DEBUG] Link: /usr/bin/arm-none-eabi-gcc @./BUILD/NUCLEO_F746ZG/GCC_ARM/.link_files.txt
[DEBUG] Return: 1
[DEBUG] Errors: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o): Conflicting CPU architectures 13/1
[DEBUG] Errors: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o)
[DEBUG] Errors: collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/home/sc/dev/projects/ssp_ethernet_transmitter/src/mbed-os/tools/make.py", line 281, in <module>
    toolchain))
  File "/home/sc/dev/projects/ssp_ethernet_transmitter/src/mbed-os/tools/build_api.py", line 534, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "/home/sc/dev/projects/ssp_ethernet_transmitter/src/mbed-os/tools/toolchains/__init__.py", line 1029, in link_program
    self.link(elf, r.objects, r.libraries, r.lib_dirs, r.linker_script)
  File "/home/sc/dev/projects/ssp_ethernet_transmitter/src/mbed-os/tools/hooks.py", line 34, in wrapper
    res = function(t_self, *args, **kwargs)
  File "/home/sc/dev/projects/ssp_ethernet_transmitter/src/mbed-os/tools/toolchains/gcc.py", line 249, in link
    self.default_cmd(cmd)
  File "/home/sc/dev/projects/ssp_ethernet_transmitter/src/mbed-os/tools/toolchains/__init__.py", line 1057, in default_cmd
    raise ToolException(_stderr)
ToolException: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o): Conflicting CPU architectures 13/1
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o)
collect2: error: ld returned 1 exit status
```

**Steps to reproduce**
* Create main.cpp file with int main() { return 0; }
* run mbed compile -t GCC_ARM -m NUCLEO_F746ZG

bcostm, 2017-06-01:
I think your GCC compiler is too old to support this CPU.  I have the version 5.4 2016q3 and the compilation is OK on my side.

scamille, 2017-06-01:
Updating to GCC6 for arm-none-eabi using https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa PPA resolved the Issue.

Thanks!

closed 2017-06-01 by scamille