ARMmbed/mbed-os #4429: ATCmdParser() enhancements

RobMeades, 2017-06-02:
### Description
- Type: Enhancement
- Priority: Minor

## Enhancement

#4119 introduce the new ATCmdParser(), which is very useful indeed.  However it's lacking some stuff to make it properly usable.  These stuffs are:

1. A method that simply allows URCs to run. Often all the application has to do is wait for the modem to spit out a URC saying it's got something, which will be trapped by a `oob()`.  However `oob()`s are only processed when a `recv()` call is in progress. My workaround, at the moment is to do `recv("\x01")`, since I know the modem will never send "\x01", but that is a somewhat peculiar hack and looks very strange in terms of the debug prints (smiley faces and timeouts appear).  It would be better to have a specific method for this, maybe with its own personal timeout.  Something like `wait(int timeout)`.
2. A method to remove an `oob()`.  Currently they can only be added, and many may be taking up RAM unnecessarily.
3. The `read()` method reads an exact number of characters from the serial stream, or -1 if there is a timeout.  The problem with this is that it might have read, say, 99 out of 100 characters from the serial stream and then timed out.  When -1 is returned the application has no idea that there really have been some characters received, so those are effectively lost.  It would be *much* better if `read()` simply returned the number of characters read, 0 representing the timeout case.
4. This is an optional one: since there can be many `oob()`s, they can take up a lot of processing time during the `recv()` call.  If there is a lot of data coming in, this could result in data loss if there is no flow control on the incoming stream.  So it would be nice to have an option to not allow `oob()`s to run on `recv()` calls where timing is critical.  This could be done as an additional bool parameter to `recv()`, which defaults to allowing `oob()`s to run.

Rob

0xc0170, 2017-06-05:
cc @hasnainvirk @kjbracey-arm

AriParkkila, 2018-06-18:
@0xc0170 please close this issue, see internal issue MBOTRIAGE-329.

closed 2018-06-18 by 0xc0170