ARMmbed/mbed-os #4432: UBLOX EVK ODIN W2 with LWIP6 ethernet fails on eth.connect() with error -3010

teetak01, 2017-06-02:
- Type: Bug
- Priority: Critical

Ublox board reliable fails to get IPv6 address with LWIP6 enabled and using easy-connect.

```
00:02:42.599 10:59:40.680 | D1 <<< DutThread: [EasyConnect] IPv6 mode
00:02:42.601 10:59:40.683 | D1 <<< DutThread: [EasyConnect] Using Ethernet
00:03:00.554 10:59:58.636 | D1 <<< DutThread: [EasyConnect] Connection to Network Failed -3010!
00:03:00.555 10:59:58.637 | TC     MainThread: Test Case fails because of: Failed to initialize connection
```

In the same network and with same application, K64F is able to connect with IPv6.

teetak01, 2017-06-02:
@MarceloSalazar @bridadan @JanneKiiskila

0xc0170, 2017-06-02:
@andreaslarssonublox

JanneKiiskila, 2017-06-02:
easy-connect is public, of course.
https://github.com/ARMmbed/easy-connect
-> using the tip of the master from there.

mbed-os - 5.4.6 release.

JanneKiiskila, 2017-06-02:
Hi,

You can use for example mbed-os-example-client for reproducing this.

https://github.com/ARMmbed/mbed-os-example-client

andreaslarssonublox, 2017-06-02:
@JanneKiiskila 
Ok, thanks for the error report we will look at shortly.
Internal ref: TE_OM2-61

Do you know if it works for the NUCLEO_F429ZI target?

JanneKiiskila, 2017-06-02:
Nucleo fails as well, see https://github.com/ARMmbed/mbed-os/issues/4433

teetak01, 2017-06-05:
This issue seems to be related to #4433 . The Ublox at least makes some effort with the MAC, although its randomization is very weak.

From couple of last runs:
```

MAC address d4:ca:6e:70:72:0d
MAC address d4:ca:6e:70:72:0d
MAC address d4:ca:6e:70:72:0c
MAC address d4:ca:6e:70:49:a1
MAC address d4:ca:6e:70:49:a1
```

With high probability there is going to be devices with identical MACs.

andreaslarssonublox, 2017-06-05:
@teetak01 The ethernet MAC address for ODIN is stored in OTP during production so it shall always be unique and does not change between reboots. Just to be sure, are you using only one board in the test setup above for the test runs?

JanneKiiskila, 2017-06-05:
@andreaslarssonublox  - does this also apply for the Ethernet?

andreaslarssonublox, 2017-06-05:
@JanneKiiskila Yes, it's stored in OTP for both ethernet and Wi-Fi.

teetak01, 2017-06-05:
@andreaslarssonublox you are right. We have multiple boards in CI and I went through the logs. Each device is having correctly unique MAC, which is preserved between jobs.

So, this issue is something else than what we are seeing with the Nucleo-board.

JanneKiiskila, 2017-06-06:
We would greatly appreciate any pointers on how to enable the right logs in LWiP that could reveal the true root cause for this problem.

bridadan, 2017-06-06:
There's a high probability that the fix for #4433 will also be the fix for this issue. Let's keep the conversation on #4433 for now and we'll update this one after more investigation.

mikaleppanen, 2017-06-09:
This could be also due to multicast filtering for ipv6. I'll check that.

mikaleppanen, 2017-06-09:
Made pull request: https://github.com/ARMmbed/mbed-os/pull/4513

teetak01, 2017-08-17:
Fixed as mentioned in above comments.

closed 2017-08-17 by teetak01