ARMmbed/mbed-os #4433: Nucleo F429ZI with LWIP6 ethernet fails on eth.connect() with error -3010

teetak01, 2017-06-02:
- Type: Bug
- Priority: Critical

Nucleo board reliable fails to get IPv6 address with LWIP6 enabled and using easy-connect.

```
10:45:36.473 | D1 <<< DutThread: [EasyConnect] IPv6 mode
10:45:36.474 | D1 <<< DutThread: [EasyConnect] Using Ethernet
10:45:53.207 | D1 <<< DutThread: [EasyConnect] Connection to Network Failed -3010!
10:45:53.208 | TC     MainThread: Test Case fails because of: Failed to initialize connection
```

In the same network and with same application, K64F is able to connect with IPv6.

teetak01, 2017-06-02:
@MarceloSalazar @bridadan @JanneKiiskila

0xc0170, 2017-06-02:
cc @bcostm @adustm @LMESTM @jeromecoutant

JanneKiiskila, 2017-06-02:
Hei,

some extra info. Easy-connect, using tip of master, mbed OS 5.4.6 release.
You can use mbed-os-example-client for reproducing the issue.

https://github.com/ARMmbed/mbed-os-example-client

That already has the same things in place.

bridadan, 2017-06-02:
@teetak01 We have seen similar issues. Do you have multiple Nucleo F429ZI devices running network tests at once? All Nucleos 429ZI devices have the **same MAC address**, so it can fail to get an address this way.

teetak01, 2017-06-05:
@bridadan Yes we do, multiple devices. Should something like this be implemented for every board if the Hardware does not provide a unique one? https://github.com/ARMmbed/mbed-os/pull/550

JanneKiiskila, 2017-06-05:
Yes, this is a serious problem and really, we should allow any board that doesn't support unique MAC address to be a reference for anything, really. It's not that a difficult thing to implement and one can use for example the unique chip id as the root for that.

jeromecoutant, 2017-06-05:
Hi
Note that ST doesn't provide any MAC address for any boards.
You have to code it.

JanneKiiskila, 2017-06-05:
Why?

SeppoTakalo, 2017-06-05:
I'm not sure that I'm looking at the correct location, but for me it looks like the functionality is already there:

https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/stm32xx_emac.c#L480
https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/stm32xx_emac.c#L498

Similar to NXP boards (K66F and K64F):
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/mbed_overrides.c#L45

Both seems to be using chip's unique ID reqisters to generate MAC addresses for testing purposes. Why this doesn't work, is beyond me...

jeromecoutant, 2017-06-05:
Yes, MAC coding is already there.

@bridadan 
> All Nucleos 429ZI devices have the same MAC address

This is not normal.
Is it possible to display this MAC address in one test?
And ideally add this kind of code:

uint32_t ST_UID[3] = {0};
HAL_GetUID(ST_UID);
printf("Unique ID %08x %08x %08x\n", ST_UID[0], ST_UID[1], ST_UID[2]);

teetak01, 2017-06-05:
It most likely is not MAC issue. I checked our logs from couple of last days:

Board | MAC
------------ | -------------
079602210B2668133462FEB8 | 00:80:e1:2b:00:20
079602210B2668133462FEB8 |  00:80:e1:2b:00:20
079602210B2068073464FEAC |  00:80:e1:2d:00:34
079602210B3368093477FEA2 | 00:80:e1:3e:00:3a
079602210B3368093477FEA2 | 00:80:e1:3e:00:3a

So a specific board always has its own unique ID.

bridadan, 2017-06-05:
I ran some tests locally, turns out I was mistaken, thanks for not taking my word for it @teetak01!

Did this bug start appearing on 5.4.6? Was it ok with 5.4.5?

teetak01, 2017-06-05:
I have not tested with much older code. We recently enabled IPv6 tests with this board.

bridadan, 2017-06-05:
We unfortunately do not have an IPV6 network. The best we have is a test router that can assign IPV6 addresses. After a few tests, it seems like the NUCLEO board can get IPV6 addresses. How often does this happen? We've seen similar issues with this board on IPV4 in the Austin CI, so we'd definitely like to track down the problem!

teetak01, 2017-06-06:
I our network, it happens every single time. Never seen IPv6 connect to pass on this board. K64F has no issues.

JanneKiiskila, 2017-06-06:
Can someone give us some tips what LWIP or other logs we should enable in mbed OS and HOW to see where the exact point of failure is?

bridadan, 2017-06-06:
@teetak01 We were able to reproduce the issue on your network using your test setup. @geky did some investigation yesterday, not sure the current status. Maybe he can provide some details?

geky, 2017-06-06:
Sorry about not leaving an update yesterday, we were able to reproduce the issue pretty consistently on the Oulu network, though we can't reproduce the issue locally.

It's a bit hacky at the moment (needs to be moved into a config file) but you can enable lwip tracing by enabling LWIP_DEBUG ([here](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwipopts.h#L71)), and switching on debug for the features you're looking at ([here](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwipopts.h#L209-L240)). In this case I turned on DHCP_DEBUG and ETHARP_DEBUG (I've attached the log [nucleo_f429zi-dchp.txt](https://github.com/ARMmbed/mbed-os/files/1055687/nucleo_f429zi-dchp.txt)).

Unfortunately it seems the device simply isn't getting any response back from a dhcp server. Running the same program on a K64F seems to have similar traffic (and same network?) but does get a response from a dhcp server. This means either the dhcp server is rejecting the request for some reason, or the nucleo is dropping packets enough to never get a dhcp response (although why does ipv4 work in this case?).

One very strange point to note from the log is the abundance of whatever the hell these 8899 broadcast packets are:
```
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:58:8b:f3:fb:03:61, type:8899 // ?
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:04:bf:6d:26:8d:59, type:8899 // ?
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:58:8b:f3:fb:01:45, type:8899 // ?
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:ba:42:b2:73:53:b4, type:800  // ipv4
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:80:2a:a8:8f:96:3c, type:806  // arp
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:04:bf:6d:26:8d:45, type:8899 // ?
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:04:bf:6d:26:8d:c0, type:8899 // ?
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:58:8b:f3:fb:01:58, type:8899 // ?
```

Googling around it seems like these come from a sort of remote switch management protocol for Realtek:
https://en.wikipedia.org/wiki/Realtek_Remote_Control_Protocol

Why these are being broadcasted to every device I'm not sure, but it is conceivable that the number of rrcp packets is saturating the little nucleo's ethernet buffers (there's a lot of these packets). That still raises the questions of why does ipv4 work, so this may just be a red herring, but I'm going to see if I can create the same heavy broadcast conditions locally. I'll let you know what happens

geky, 2017-06-06:
Ok, so the rrcp doesn't seem to be the problem, I created similar traffic locally and the NUCLEO_F429ZI didn't have any issues.

We also tried a different mac addresses without success.

I'm not really sure what the issue is, somewhere dhcp packets just aren't getting to the nucleo. At this point the best way to progress would be to create a log of ethernet traffic, that would quickly narrow down the problem. Puting the nucleo under a seperate switch may also fix the problem, and that would narrow it down to other traffic interfering (although that may raise more questions than answers).

jeromecoutant, 2017-06-09:
Hi
This issue looks corrected with #4513 
Thx @mikaleppanen

geky, 2017-06-09:
Ah! Very cool and good to know, thanks @mikaleppanen

bridadan, 2017-06-12:
Good to close now that #4513 is merged?

teetak01, 2017-06-12:
Thanks for all!

closed 2017-06-12 by teetak01