ARMmbed/mbed-os #4440: Advance notice of folder structure change in under "/targets/TARGET_RENESAS"

TomoYamanaka, 2017-06-05:
@mbedNoobNinja

Hello, we are Renesas mbed team.

We will release a new mbed board that mounts RZ/A1.
Related to that, we consider that would like to share common codes in the code below HAL in mbed-os.
The structure that we would like to change is under `/targets/TARGET_RENESAS`, we show its structure in the below.

New folder structure will be merged by end of Sep.
So, please let me know if there is any problem about new folder structure by end of Aug.
Also, we would appreciate it if you could execute the test by using new folder structure.
<br>

**Current state:**
The folder are separated by mbed board unit, and there are many same files in these folders.
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_RENESAS

**proposed change:**
The same files are shared, only difference files are separated in folder by mbed board unit.
In this way, maintenance and new board supporting will be very easy.
https://github.com/TomoYamanaka/mbed/tree/Feature_LYCHEE/targets/TARGET_RENESAS

theotherjimmy, 2017-06-05:
@TomoYamanaka  That seems fine. I don't forseee the build system having any problems with that.

Would you also like an advance review of your changes to targets.json?

theotherjimmy, 2017-06-05:
https://github.com/ARMmbed/mbed-os/compare/master...TomoYamanaka:Feature_LYCHEE That's the diff with master. Your branch seems to be a bit behind, and it would not be accepted if you submitted a PR right now.

TomoYamanaka, 2017-06-06:
@theotherjimmy 

Thank you for your comments.

> Would you also like an advance review of your changes to targets.json?

Thank you. 
Please let me know if there is any problem.
I will reflect its comments when I PR.

> That's the diff with master. Your branch seems to be a bit behind, and it would not be accepted if you submitted a PR right now.

That's right. I understand that this branch is not up to date.
This branch is based on mbed-os-5.4.3 for some development.
For now, I will PR by end of Aug. At that time I will PR this branch after merge the latest master.

mbedNoobNinja, 2017-06-06:
@TomoYamanaka  that's great, we are glad you are considering that.

> Also, we would appreciate it if you could execute the test by using new folder 

We will be able to run the tests not sooner than 2-3 weeks.

TomoYamanaka, 2017-06-06:
@mbedNoobNinja 

Thank you for your comments.

> We will be able to run the tests not sooner than 2-3 weeks.

We are looking forward to your test result.
<br><br>

By the way, 
Do you know that Cortex-A do **not support in mbed OS 5.5** which will be released in end of June?
Cortex-A **will be supported in mbed OS 5.6**. But until then, mbed boards of cortex-A need to use mbed OS 5.4.
Result of discussion with ARM mbed team, we agreed with them to output the below build error messages in non-support period.
- Could not compile for RZ_A1H:
- RZ_A1H Will be supported in mbed OS 5.6. 
- To use the RZ_A1H, please checkout the mbed OS 5.4 release branch. 
- See https://developer.mbed.org/platforms/Renesas-GR-PEACH/#important-notice for more information.

In case of VK-RZ/A1H, the part of RZ_A1H will be replaced by VK-RZ/A1H.

In the above URL, we will guide the way to checkout mbed OS 5.4.
The guide example is here : 
https://developer.mbed.org/users/HinoNaka/notebook/renesas-gr-peach/
chapter "Important Notice"

Although this guide URL is in GR-PEACH(RZ_A1H) page, can we mention about VK_RZ_A1H at this URL?

TomoYamanaka, 2017-06-09:
@mbedNoobNinja 

Thank you.
We are looking forward to your reply.

mbedNoobNinja, 2017-06-27:
Finally ...
Sorry for the delay here are the results as were promised:
**[cli_tst_summary.txt](https://github.com/ARMmbed/mbed-os/files/1105221/cli_tst_summary.txt)**
**[old_tst_summary.txt](https://github.com/ARMmbed/mbed-os/files/1105222/old_tst_summary.txt)**
here are the complete logs:
[old_tst_arm_iar_gcc.txt](https://github.com/ARMmbed/mbed-os/files/1105241/old_tst_arm_iar_gcc.txt)
[cli_tst_arm.txt](https://github.com/ARMmbed/mbed-os/files/1105242/cli_tst_arm.txt)
[cli_tst_gcc.txt](https://github.com/ARMmbed/mbed-os/files/1105243/cli_tst_gcc.txt)
[cli_tst_iar.txt](https://github.com/ARMmbed/mbed-os/files/1105244/cli_tst_iar.txt)

Not bad, except the IAR ones ...

theotherjimmy, 2017-06-27:
@TomoYamanaka @mbedNoobNinja The branch *Feature_LYCHEE* seems to be based on a release branch. We do not accept PRs to release branches at the moment, so I recommend that you base your work on master.

mbedNoobNinja, 2017-06-28:
... Well the error was in my TV! (more specifically in execution of **switch** command of the TortoiseGit) In attempt to **_switch_** from **master** to **Feature_LYCHEE** that mess up the whole project and cause the strange python issue. After fresh & clean checkout of the Feature_LYCHEE branch with TortoiseGit command (Git Clone) there were no python errors anymore.
... That's fine, but new issue come up:
`[Error] i2c_api.c@64,0:  #154: expression must have struct or union type`
`[Error] spi_api.c@47,0:  #154: expression must have struct or union type`
`[Error] serial_api.c@151,0:  #136: struct "serial_s" has no field "serial"`
...Anyway after adding **"I2C_ASYNCH"**, **"SPI_ASYNCH"** & **"SERIAL_ASYNCH"** interfaces as "device_has" ittems for the _"VK_RZ_A1H"_ platform in **targets.json**, the errors disappeared and the test finally managed to run.

Here are the results:
[xx_cli_tst_summary.txt](https://github.com/ARMmbed/mbed-os/files/1109584/xx_cli_tst_summary.txt)
[xx_old_tst_summary.txt](https://github.com/ARMmbed/mbed-os/files/1109583/xx_old_tst_summary.txt)
and the logs:
[xx_cli_tst_arm.txt](https://github.com/ARMmbed/mbed-os/files/1109589/xx_cli_tst_arm.txt)
[xx_cli_tst_gcc.txt](https://github.com/ARMmbed/mbed-os/files/1109588/xx_cli_tst_gcc.txt)
[xx_cli_tst_iar.txt](https://github.com/ARMmbed/mbed-os/files/1109590/xx_cli_tst_iar.txt)
[xx_old_tst_arm_iar_gcc.txt](https://github.com/ARMmbed/mbed-os/files/1109587/xx_old_tst_arm_iar_gcc.txt)

Definitely there is some issue with the IAR binaries, it looks they stuck somewhere.
The other total failure is `features-feature_lwip-tests-mbedmicro-net-tcp_hello_world`

That's the situation for now ... any help will be welcome!

TomoYamanaka, 2017-06-29:
@mbedNoobNinja 

Thank you for your comments, and I appreciate that you execute the test.

Is it right that the rest problems is the below?
> Definitely there is some issue with the IAR binaries, it looks they stuck somewhere.
> The other total failure is  features-feature_lwip-tests-mbedmicro-net-tcp_hello_world 

This Feature_LYCHEE branch is based on the release branch of [mbed-os-5.4.3](https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-5.4.3).
So, can you execute the test again by using mbed-os-5.4.3?

TomoYamanaka, 2017-06-29:
@theotherjimmy

Thank you for your comments.

> @TomoYamanaka @mbedNoobNinja The branch Feature_LYCHEE seems to be based on a release branch. We do not accept PRs to release branches at the moment, so I recommend that you base your work on master.

That's right.
This branch is based on release branch of mbed-os-5.4.3 for advance development.
Therefore, I will PR, after I merge Feature_LYCHEE branch to the latest master branch in advance.

mbedNoobNinja, 2017-06-29:
Isn't it 5.4.3 already, if it's based on it ?. I don't know how to explicitly checkout mbed-os-5.4.3.
The tags in the branch Feature_LYCHEE are:
```
feature_LYCHEE_rev1
feature_LYCHEE_rev2
feature_LYCHEE_rev3
mbed-os-5.1.0
mbed-os-5.1.0-rc1
mbed-os-5.1.0-rc2
mbed-os-5.1.0-rc3
mbed-os-5.1.0-rc4
mbed-os-5.1.0-rc5
mbed-os-5.2.0-rc1
mbed-os-5.3.0-rc1
mbed-os-5.4.0
mbed-os-5.4.0-rc1
mbed-os-5.4.0-rc2
mbed-os-5.4.1
mbed-os-5.4.2
mbed_lib_rev100
mbed_lib_rev101
mbed_lib_rev102
mbed_lib_rev103
mbed_lib_rev104
mbed_lib_rev105
mbed_lib_rev106
mbed_lib_rev107
mbed_lib_rev108
mbed_lib_rev110
mbed_lib_rev111
mbed_lib_rev112
mbed_lib_rev113
mbed_lib_rev114
mbed_lib_rev116
mbed_lib_rev117
mbed_lib_rev118
mbed_lib_rev119
mbed_lib_rev120
mbed_lib_rev121
mbed_lib_rev122
mbed_lib_rev138
mbed_lib_rev139
mbed_lib_rev65
mbed_lib_rev67
mbed_lib_rev68
mbed_lib_rev69
mbed_lib_rev70
mbed_lib_rev71
mbed_lib_rev73
mbed_lib_rev74
mbed_lib_rev75
mbed_lib_rev76
mbed_lib_rev77
mbed_lib_rev78
mbed_lib_rev79
mbed_lib_rev80
mbed_lib_rev81
mbed_lib_rev82
mbed_lib_rev83
mbed_lib_rev84
mbed_lib_rev85
mbed_lib_rev86
mbed_lib_rev87
mbed_lib_rev88
mbed_lib_rev89
mbed_lib_rev90
mbed_lib_rev91
mbed_lib_rev92
mbed_lib_rev93
mbed_lib_rev94
mbed_lib_rev95
mbed_lib_rev96
mbed_lib_rev97
mbed_lib_rev98
mbed_lib_rev99
mbed_liv_rev108
mbed_private_fsl_initial
```
There is no `5.4.3` version in it.
From master's branch point of view there is tag `mbed-os-5.4.3` and when I checkout it the folder structure isn't combined as in Feature_LYCHEE (I mean there is no such folder **TARGET_RZA1XX**) Should I run the test no matter what the folder structure is ?

theotherjimmy, 2017-06-29:
> I will PR, after I merge Feature_LYCHEE branch to the latest master branch in advance.

Please do not merge. There are several reasons not to:
 * We do not accept PRs to master with merge commits
 * You would be introducing duplicate history (master's and the release branch's)

Instead, you will have to cherry pick the commit range. You cannot rebase because there are duplicated commits on the release branch with different hashes, because they have a different ancestor. If you were to rebase, git would try to apply all of the release commits onto master in your branch. This is why I recomend doing development off master whenever possible: it allows you to rebase to bring things up to date.

TomoYamanaka, 2017-06-30:
@mbedNoobNinja

Thank you for test of mbed OS 5.4.3.
Sorry for lack of description.
Because I wanted to comfirm whether the new folder structure has affected, I asked for test to you.

As far as the test result, it seems that `features-feature_lwip-tests-mbedmicro-net-tcp_hello_world` test is fail in mbed OS 5.4.3 too.
I was relieved that the new folder structure did not seem to be affected.

TomoYamanaka, 2017-06-30:
@theotherjimmy

Thank you for your comments.

I am going to process as the follow.
1. I rebase my master branch to latest.
2. I merge only the code changing by Feature_LYCHEE branch to my master branch.
3. I PR from my master branch.

Could you give me your opinion about the above?

theotherjimmy, 2017-06-30:
@TomoYamanaka Sure. I'm happy to provide my opinion. There are two options depending on what you mean bay "latest". If you mean the tag "latest" then you will experience the same pitfalls I mentioned above, but with a more recent release branch. The "latest" tag points to the most recent release. If you meant "latest" to mean the most recent version of the master branch, then that sounds great (I have another recommendation, just a sec), but please avoid using the word "merge" when describing the process. In reality the git command that you would end up using for this process is either `git rebase` or `git cherry-pick`. I recommend, the following process, a variation of yours:
 
 1) Pull master from ARMmbed `git checkout master; git pull armmbed master`
 2) Checkout new branch with the same commit a master `git checkout -b add-lychee`
 3) Rebase/Cherry-pick the commits from the Feature_LYCHEE branch `git rebase --onto master add-lychee Feature_LYCHEE`
 4) PR from your add-lychee

The advantage of this proccess, is that it leaves your master branch pristine. That would allow you to work on other PRs in the mean time.

TomoYamanaka, 2017-07-03:
@theotherjimmy

Thank you for your comments.
Soory fot lack of description.
The "latest" I mentioned is that the most recent version of the master branch when I PR.

And thank you for advicing too regard to describing the process.

theotherjimmy, 2017-07-03:
@TomoYamanaka You're welcome. Please let me know if you run into any issues when brining the branch up to the latest version of master.

closed 2017-07-20 by TomoYamanaka