ARMmbed/mbed-os #4442: Multicast group leave and join socket opts are unusable with mbed-mesh-api.

tommikas, 2017-06-05:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

The Nanostack socket option to join or leave multicast groups requires Nanostack's internal network interface ID. This is hidden from the application logic inside mbed-mesh-api so these socket options are unusable using the netsocket API.

tommikas, 2017-06-05:
I don't think this is quite a duplicate of #4034 but related in that if an abstraction for common socket options such as suggested there was implemented, these could perhaps be included in the list.

0xc0170, 2017-06-05:
cc @SeppoTakalo

SeppoTakalo, 2017-06-05:
Well.. I feel like this is a duplicate of the linked issue.
There is no portable socket options in mbed OS.

tommikas, 2017-06-05:
That issue is more about having to use the stack specific API directly, which is how it was intended to be used when it comes to socket options so it's mostly a non-issue. What @karsev wanted to do could easily be achieved with the APIs as they are.

This is a bit different in that these specific options simply can't be used because the stack abstraction in mbed-mesh-api makes it impossible by hiding the interface ID.

Arguing about whether it's a duplicate is pointless though. Of course a generic solution would be better than just doing it in mbed-mesh-api for Nanostack.

tommikas, 2017-06-05:
[From the linked issue](https://github.com/ARMmbed/mbed-os/issues/4034#issuecomment-289512203): 
>I'm not sure it's possible to create an exhaustive list of network options, so allowing stack-specific options lets specific network stacks support features without requiring the features from other network stacks.

This issue with these socket options exists with Nanostack, but would it necessarily exist with other network stacks?

tommikas, 2017-06-26:
Closing this as there is a way to use these options without the interface ID after all.

As [documented for the SOCKET_IPV6_MULTICAST_IF socket option](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/FEATURE_NANOSTACK/sal-stack-nanostack/nanostack/socket_api.h#L745-L746) the interface id 0 can be used for an unspecified interface.

closed 2017-06-26 by tommikas