ARMmbed/mbed-os #4443: Implement recvmsg support for netsocket API

tommikas, 2017-06-05:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement
- Priority: Major

---------------------------------------------------------------
## Enhancement

Without support for `socket_recvmsg` for example resolving the destination address of a received UDP packet is not possible. This is needed when receiving from multiple multicast groups on a single socket.

Currently at least nanostack provides `socket_recvmsg`.

0xc0170, 2017-06-05:
cc @kjbracey-arm @geky

geky, 2017-06-12:
Looks like you're right, we don't currently have a method for determining the source address of udp packets. The recvmsg function would work and will need to be added to the UDPSocket and NetworkStack classes along with testing and documentation. I don't believe any one is currently working on this.

kjbracey-arm, 2017-06-14:
My general recommendation is to distinguish the services you want to talk to by stuff that is easily visible - port numbers and URIs.

Whatever service you're providing (eg light control) should presumably also be accessible via unicast as well as multicast - so for the unicast case you'd be inspecting payload and port to determine what service was being asked for - you couldn't use the destination address.

tommikas, 2017-06-15:
Our intention wasn't to distinguish the service by the multicast group. The idea was that for example in the case of a light control service there may be several "control groups" which can be mapped to different multicast groups. Each node joins the multicast groups that correspond to the control groups it belongs to.

That said, it might be simpler to separate the "split the network into subnets" and "distinguish between control groups" problems. In the first case the destination address isn't needed, and for the latter parsing the payload could do. @mikter Would that be good enough for you?

SeppoTakalo, 2018-06-18:
The POSIX version of `recvmsg()` does not actually give the destination address of incoming packet.
http://pubs.opengroup.org/onlinepubs/9699919799/functions/recvmsg.html
```
ssize_t recvmsg(int socket, struct msghdr *message, int flags);
...
message
  Points to a msghdr structure, containing both the buffer to store the source address and the
  buffers for the incoming message. The length and format of the address depend on the address
  family of the socket.
```

There is a way to do this in Linux, but when you need API that standard does not provide, you might want to consider that do you need to redesign your application in order to make it portable.

For example, if you seriously need to use different groups, why does the group name/id/number do not exist in the query packet?

I would like to close this request as "Won't fix", I don't really see the usecase. If it cannot be done in POSIX, you have flaw in your design.

For a background information, here is discussion about this in Linux: https://stackoverflow.com/questions/5281409/get-destination-address-of-a-received-udp-packet

closed 2018-10-02 by adbridge