ARMmbed/mbed-os #4458: mbed OS support USBDevice feature on nRF52840 ?

andcor02, 2017-06-06:
This is a common recurring question, what are the plans for USBDevice for devices and mbed os.

Specifically the nRF52840 mbed enabled board, is this available in the future for USBDevice?

This has been flagged here: https://developer.mbed.org/questions/78204/Will-the-mbed-OS-support-USBDevice-featu/

Regards,

Andrea

xuio, 2017-09-24:
what is the status on this issue? Is there any new development yet?

kubaraczkowski, 2018-04-04:
Hi,
Now that NRF SDK 15 brings "production quality" USB drivers, perhaps this issue could be addressed?

0xc0170, 2018-04-04:
cc @marcuschangarm @c1728p9

marcuschangarm, 2018-04-04:
There is a new Mbed USB HAL scheduled for 5.9 so we wont be using much of the Nordic USB drivers.

kubaraczkowski, 2018-04-04:
Souds great!
For when is 5.9 scheduled?

AGlass0fMilk, 2019-01-14:
I am currently working on implementing the USBDevice HAL on the nRF52840, see #7674 

It may be stable enough for you to mess around with but the test suite is very thorough so it’s not passing all the Mbed tests yet.

linlingao, 2019-05-30:
This is duplicate of #7674. I'm closing this issue. Please track under #7674.

closed 2019-05-30 by linlingao