ARMmbed/mbed-os #4460: project.py - the files are missing in the exported project

0xc0170, 2017-06-06:
### Description
- Type: Bug 
- Priority: Major

Running
```
python tools\project.py -m NUCLEO_F411RE -i uvision -p 48
```

Results in 2 projects (one zipped, one in the build folder). None of these 2 projects build however. The first in the root has wrong file paths . In this case it is for uvision. The second zipped project has paths correctly set, but cant find all the header files (receiving error core_cm4.h not found, it is not in the zipped archive).

I believe I reported this earlier but could not find open/closed issue related to this one.

cc @theotherjimmy

theotherjimmy, 2017-06-06:
@0xc0170 exporting of tests is not currently supported. I will submit a PR to disable those flags.

OTOH, You have to remember that tests do not compile with the root of mbed. They have there own series of paths.

0xc0170, 2017-06-06:
How to debug the test then ? :-)

theotherjimmy, 2017-06-06:
Build with the debug profile? and use the elf? Maybe create a simple example program and debug that?

closed 2018-03-28 by ChiefBureaucraticOfficer