ARMmbed/mbed-os #4461: Realtek Ameba not implementing FlashIAP

JanneKiiskila, 2017-06-06:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
RealTek Ameba - 
mbed compile -m REALTEK_RTL8195AM 

**Toolchain:**
N/A

**Toolchain version:**
N/A

**mbed-cli version:** 
(`mbed --version`)
1.1

**meed-os sha:**
(`git log -n1 --oneline`)
mbed-os-5.5-rc1
commit 92fbf2a9b3988d430482fc25a6077f2462e2a634 (HEAD -> mbed-os-5.5, tag: mbed-os-5.5.0-rc1, origin/mbed-os-5.5)

**DAPLink version:**
N/A

**Expected behavior**

FlashIAP is implemented.

**Actual behavior**

FlashIAP is not implemented, therefore cloud client can't compile on Realtek Ameba.

**Steps to reproduce**

Try to compile the mbed-cloud-client-example using mbed OS 5.5. RC1 and current tip of mbed-client-pal.

Or create minimal program that tries to use `FlashIAP flash;`

JanneKiiskila, 2017-06-06:
@MarceloSalazar @avolinski @marcuschangarm @0xc0170 - please note.

0xc0170, 2017-06-06:
cc @Archcady - I could not locate the flash IAP on the branch, was it implemented?

screamerbg, 2017-07-17:
@JanneKiiskila is this still relevant? Seems FLASH HAL is implemented https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/flash_api.c. Please close if completed.

JanneKiiskila, 2017-07-17:
Yes, this can be closed. **We haven't tested it though,** because the linker issue is still unresolved.

closed 2017-07-17 by JanneKiiskila