ARMmbed/mbed-os #4462: Realtek Ameba not linking

JanneKiiskila, 2017-06-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Realtek Aemba

**Toolchain:**
GCC_ARM

**Toolchain version:**
```
~/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

**mbed-cli version:** 
(`mbed --version`)
mbed --version
1.1.1

**meed-os sha:**
(`git log -n1 --oneline`)
92fbf2a (HEAD -> mbed-os-5.5, tag: mbed-os-5.5.0-rc1, origin/mbed-os-5.5) Add versioning information to mbed OS 5.5 release branch

**DAPLink version:**
N/A

**Expected behavior**

It should link the Cloud Client w/o update (to remove FlashIAP).

**Actual behavior**

Linking fails.

```
Link: mbed-cloud-client-example-internal
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: ./BUILD/REALTEK_RTL8195AM/GCC_ARM/mbed-cloud-client-example-internal.elf section `.bss' will not fit in region `BD_RAM'
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region RAM exceeds ram limit
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `BD_RAM' overflowed by 16352 bytes
collect2: error: ld returned 1 exit status
[ERROR] /home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: ./BUILD/REALTEK_RTL8195AM/GCC_ARM/mbed-cloud-client-example-internal.elf section `.bss' will not fit in region `BD_RAM'
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region RAM exceeds ram limit
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `BD_RAM' overflowed by 16352 bytes
collect2: error: ld returned 1 exit status
```

**Steps to reproduce**

Compile and link mbed Cloud Client example (available only to Cloud Client Partners, sorry).

JanneKiiskila, 2017-06-06:
@0xc0170 @avolinski @yogpan01 - please note this one as well, next hurdle then...

0xc0170, 2017-06-06:
cc @Archcady

Archcady, 2017-06-07:
I think we didn't switch on our TCM section so it does not have much space for user code. The possible solution for this is to switch on TCM usage.

JanneKiiskila, 2017-06-07:
Whatever it takes!

Archcady, 2017-06-07:
Hi @JanneKiiskila  I switch on the TCM usage here #4487 in order to resolve this issue. Please try the test again, thank you.

JanneKiiskila, 2017-06-12:
Hei,

as it was now merged to master, it was easier for me to compile it (last week was too busy), but it still unfortunately fails. 

mbed-os master was now pointing to: f31ea012378ba7fd26429a805e8cf5cb4999fb25

So it had the fix in it. But, unfortunately 

```
Link: mbed-cloud-client-example-internal
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: ./BUILD/REALTEK_RTL8195AM/GCC_ARM/mbed-cloud-client-example-internal.elf section `.bss' will not fit in region `BD_RAM'
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region RAM exceeds ram limit
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `BD_RAM' overflowed by 2464 bytes
collect2: error: ld returned 1 exit status
[ERROR] /home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: ./BUILD/REALTEK_RTL8195AM/GCC_ARM/mbed-cloud-client-example-internal.elf section `.bss' will not fit in region `BD_RAM'
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region RAM exceeds ram limit
/home/jankii01/opt/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `BD_RAM' overflowed by 2464 bytes
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/jankii01/mbed/mbed-cloud-client-example-internal/mbed-os/tools/make.py -t GCC_ARM -m REALTEK_RTL8195AM --source . --build ./BUILD/REALTEK_RTL8195AM/GCC_ARM -DRESET_STORAGE" in "/home/jankii01/mbed/mbed-cloud-client-example-internal"
```

yogpan01, 2017-06-13:
Hi @Archcady ,
While compiling for mbed-os-example-client for mbedOS 5,5 (https://github.com/ARMmbed/mbed-os-example-client/tree/mbedos55-oob) , the linker is failing for ARMCC with following message
```
mbed-os-example-client$ mbed compile -m REALTEK_RTL8195AM -t ARM
Building project mbed-os-example-client (REALTEK_RTL8195AM, ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_STORAGE
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: env
Scan: mbed
Link: mbed-os-example-client
"./mbed-os/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/rtl8195a.sct", line 8 (column 2): Warning: L6312W: Empty Execution region description for region _ROM_CODE
"./mbed-os/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/rtl8195a.sct", line 44 (column 2): Warning: L6312W: Empty Execution region description for region _DRAM_CODE
Error: L6220E: Load region LR_RAM size (496632 bytes) exceeds limit (458751 bytes). Region contains 542 bytes of padding and 960 bytes of veneers (total 1502 bytes of linker generated content).
Finished: 0 information, 2 warning and 1 error messages.
[ERROR] "./mbed-os/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/rtl8195a.sct", line 8 (column 2): Warning: L6312W: Empty Execution region description for region _ROM_CODE
"./mbed-os/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/rtl8195a.sct", line 44 (column 2): Warning: L6312W: Empty Execution region description for region _DRAM_CODE
Error: L6220E: Load region LR_RAM size (496632 bytes) exceeds limit (458751 bytes). Region contains 542 bytes of padding and 960 bytes of veneers (total 1502 bytes of linker generated content).
Finished: 0 information, 2 warning and 1 error messages.

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/yogpan01/mbed-os-example-client/mbed-os/tools/make.py -t ARM -m REALTEK_RTL8195AM --source . --build ./BUILD/REALTEK_RTL8195AM/ARM" in "/home/yogpan01/mbed-os-example-client"
---
```

I think there is room to increase the limit even more since there is quite lot of unused user space left on the device.

Archcady, 2017-06-13:
Seems like we'll have to move code to SDRAM, need some time to work on this and also test this.

Archcady, 2017-06-21:
Hi @0xc0170,  we meet a problem that if we need to move code to SDRAM, then we wanna use two linking script files (.sct / .ld / .icf) which produce two images. And we wonder if this is supported by mbed-os tools. Thanks.

0xc0170, 2017-06-21:
> Hi @0xc0170, we meet a problem that if we need to move code to SDRAM, then we wanna use two linking script files (.sct / .ld / .icf) which produce two images. And we wonder if this is supported by mbed-os tools. Thanks.

Can you describe the problem in own separete issue? that would help.

JanneKiiskila, 2017-06-26:
@theotherjimmy  - can you give a hand here?

theotherjimmy, 2017-06-26:
How would you like me to help out?

JanneKiiskila, 2017-06-26:
@theotherjimmy - Do the mbed tools support 2 linking sections or not? If not, can we somehow force the SW to be put to the SDRAM only - it's 2 megabytes, so I don't see any problem in fitting everything in there.

theotherjimmy, 2017-06-26:
@JanneKiiskila The tools don't need to understand linking sections at all, and don't understand linking sections. That's up to linker scripts for the individual compilers. You would have to look at GCC_ARM, IAR and ARMC5 to determine how to write a linker script for each of them to take advantage of the different regions.

Archcady, 2017-06-29:
Sorry for the delay cos' we met some problems. Now again here's a PR trying to fix this issue: https://github.com/ARMmbed/mbed-os/pull/4665

0xc0170, 2017-06-30:
@JanneKiiskila Can you test the referenced fix above ?

JanneKiiskila, 2017-07-10:
I think the fix proposed by Archcady should go into the next mbed OS 5.5.x patch release.
That solves the linking issues. We have then more other issues, but those can't even be debugged with an official mbed OS release unless this PR (#4655) is taken in).

JanneKiiskila, 2017-07-25:
Can we get some closure this?

closed 2017-08-16 by sg-