ARMmbed/mbed-os #4486: Weird crashes caused by software_init_hook() being marked as naked.

adamgreen, 2017-06-07:
## Severity
This is blocking for anyone building mbed 2 (no RTOS) projects with GCC.

## Branch Encountered
I hit this issue when testing code from the feature_cmsis5 branch but I see the offending code is now in the **master** branch as well.

## Description
If I build the latest sources with MBED_CONF_RTOS_PRESENT undefined then I hit weird hard faults. Upon debugging the cause of the crash it turned out that the vtable pointers for objects with global constructors were still NULL (from being in .bss section). Tracing through the code I found that execution was jumping straight from software_init_hook() and into main(), skipping __libc_init_array().

Looking at the disassembly I saw:
```
(gdb) disass software_init_hook
Dump of assembler code for function software_init_hook:
   0x14000728 <+0>:	bl	0x14000724 <mbed_sdk_init>
   0x1400072c <+4>:	bl	0x14000726 <software_init_hook_rtos>
End of assembler dump.
```

There is no code in this function to save LR and use it to return upon the completion of software_init_hook_rtos() which does nothing but just return in my scenario. This happens because the function is incorrectly defined as **naked** in mbed_sdk_boot.c

```c
__attribute__((naked)) void software_init_hook(void)
71	{
72	    mbed_sdk_init();
73	    software_init_hook_rtos();
74	}
```

Removing the ```__attribute__((naked))``` fixes the issue.

0xc0170, 2017-06-07:
Thanks @adamgreen , I'll have a look at this and fix it. There was a report earlier from ST that they are hitting timeouts in the test, this might explain it.

adamgreen, 2017-06-08:
Fixed with PR #4489

closed 2017-06-08 by adamgreen