ARMmbed/mbed-os #4488: Hardcoded values for debug UART pins in TARGET_MCU_NRF52832

andreaslarssonublox, 2017-06-07:
### Description
We have development board that is based on NRF52 called UBLOX_EVK_NINA_B1. However we have different UART pins for the debug UART compared to the NRF52_DK board.
In the file sdk_config.h and nrf_drv_config.h those UART pins are hard-coded which makes it difficult for us to add our board with a working debug UART. See targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk_patch/sdk_config.h
I think that the values in PinNames.h should be used and don't know the reason why those are overridden in sdk_config.h.

- Type: Bug
- Priority: Blocker

@nvlsianpu 

---------------------------------------------------------------
## Bug

**Target**
TARGET_MCU_NRF52832

**Toolchain:**
GCC_ARM|ARM|IAR
Only tested with GCC_ARM

**Toolchain version:**
GNU Tools for ARM Embedded Processors 5.4.1

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
6b0db0d Merge pull request #4411 from hasnainvirk/add_to_namespace_mbed

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

nvlsianpu, 2017-06-07:
Thanks for description. Issue tracked internally.

andreaslarssonublox, 2017-06-15:
Hi, any news on this @nvlsianpu?

nvlsianpu, 2017-06-27:
@andreaslarssonublox  https://github.com/ARMmbed/mbed-os/pull/4647

bridadan, 2017-08-23:
Hi y'all. I was going through some old issues, looks like all of the fixes related to this one have been merged.

@andreaslarssonublox good to close?

andreaslarssonublox, 2017-08-24:
@bridadan Yes, you can close it!

closed 2017-08-24 by bridadan