ARMmbed/mbed-os #4492: How to wake up a thread when it call osdelay(oswaitforever);

kyliuxing, 2017-06-07:
## Question
I will call osdelay() to make a thread to sleep , but if i use oswaitforever as argument, how can i wake it up?

kyliuxing, 2017-06-07:
another question, I can't delete a message queue or mail queue, right?

geky, 2017-06-12:
If you want a thread to sleep while being able to be woken up your best bet is a semaphore:
```  cpp
Semaphore sema;

int thread1() {
    sema.wait(osWaitForever); // sleeps until woken up or timeout
}

int thread2() {
    sema.release(); // causes thread1 to be woken up.
}
```

Here's an example of waking up a thread on a button press:
``` cpp
InterruptIn button;
Semaphore sema;

int thread1() {
    // attach a callback to release the semaphore when the button is pressed
    button.rise(callback(&sema, &Semaphore::release));

    while (true) {
        sema.wait(); // osWaitForever is optional

        dostuffonbuttonpress();
    }
}
```

---

It looks like you're right, the mail and message queues currently don't support destruction. This may be a bug @bulislaw, @c1728p9

vargham, 2017-06-16:
Or you can wait for a signal. Documentation here: https://www.keil.com/pack/doc/CMSIS/RTOS/html/group__CMSIS__RTOS__SignalMgmt.html

Example:
```
int thread1()
{
    while (true)
    {
        osEvent event = osSignalWait(0x1, 1000); //Waits 1000 ms for signal 0x1
        if (event.status == osEventSignal)
        {
            //OK, received signal in time.
            //Clear signal
            osSignalClear(Thread::gettid(), 0x01);
            //Do something.
        }
        else if (event.status == osEventTimeout)
        {
            //Timeout, so do something else.
        }
  }
}
```

You can set the signal: `osSignalSet (tid_thread1, 0x1);`

You can use this timeout pattern with semaphores and mailboxes too.

kyliuxing, 2017-06-16:
@geky @vargham 
thanks for your help，i will try it.

closed 2017-06-16 by kyliuxing

geky, 2017-06-16:
Both should work, with thread signals being more lightweight than semaphores.

We've reserved thread signals for users, since they share a single bit-field accross the application, so I can forgetting they are available : )