ARMmbed/mbed-os #4505: OnBoardCellularInterface fails for CDMA radios.

sarahmarshy, 2017-06-08:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
MTS Dragonfly 
**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**

Device connects 

**Actual behavior**

Device fails to connect 
Here is the log: 
```shell

mbed-os-example-cellular, Connecting...
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)

Couldn't connect: -3012, will retry
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT(Timeout)
AT> AT
AT? OK%n
AT< AT
AT<
AT= OK
AT> ATE0;+CMEE=2;&K0+IPR=115200;&C1;&D0
AT? OK%n
AT<
AT< ATE0;+CMEE=2;&K0+IPR=115200;&C1;&D0
AT<
AT= OK
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)
AT> AT+CPIN?
AT? +CPIN: %*15[^
]
%n
AT<
AT! ERROR
AT(Aborted)

Authentication Failure. Exiting application

Failure. Exiting
```

**Steps to reproduce**

Use an MTS dragonfly with a CDMA radio and this repo - https://github.com/armmbed/mbed-os-example-cellular.

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

You can see in the log that we are querying for the status of the SIM PIN with CPIN? ([AT command ref](http://www.telit.com/fileadmin/user_upload/products/Downloads/2G/Telit_AT_Commands_Reference_Guide_r23.pdf)]). However, this is a CDMA radio. There is no SIM card....

**Suggested enhancement**

Support for CDMA radios. Remove this check if the device is CDMA.

hasnainvirk, 2017-06-09:
@sarahmarshy I looked at the documentation of MTQ-EV3-B01, http://www.multitech.com/documents/publications/manuals/s000627.pdf
Check Chapter 5 Carrier Specific notices. There are Carrier specific initial setups that are needed to be performed. Our implementation is based upon standard AT command set, works with MTQ-HB-B01 type of devices. Maybe a future enhancement. But out of scope currently.

sarahmarshy, 2017-06-09:
@hasnainvirk We should be very clear about what modules this driver supports. The driver should not fail like this. Can you query the device in the driver and fail if it is not supported? 

Additionally, this driver is not set up to support CDMA radios at all if it is querying a SIM pin, as they do not require SIM cards.

hasnainvirk, 2017-06-09:
@sarahmarshy Not ALL CDMA cellular modems come with a baked in SIM. Currently there are more important things are underway. I have kept a note of your issue and martin had added a label.

sarahmarshy, 2017-06-09:
@hasnainvirk Could we document this as a known issue? Or at least document the modules the driver supports?

0xc0170, 2017-09-27:
@hasnainvirk Shall we document this, or has it been added somewhere ? I am not certain this is known-issue, just a feature that is not supported yet?

hasnainvirk, 2017-09-27:
@0xc0170 Could you please assign this to someone in US ? Actually CDMA doesn't  exist in Europe. To answer your question, It has not been supported yet. Probably one way of doing this is to ask user to tell us if he has a CDMA modem with sim card baked in using json config. I am not sure if there would be a standard AT command to find out if the sim card is baked in or not. @TeemuKultala did you do any work on this aspect ?

sarahmarshy, 2017-09-27:
The current drivers by multitech use an AT command to query the module name. The driver switches to CDMA/GSM logic based on the module name.

hasnainvirk, 2017-09-28:
@sarahmarshy ATI command can give you Identification of the modem but that is very vendor specific.
This cannot really be used in a generic driver in any scalable way.

AriParkkila, 2018-06-18:
@sarahmarshy I think we can close this issue. [Mbed cellular](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/) has been redesigned to support cellular module specific adaptation also w/o SIM PIN.

closed 2018-06-18 by 0xc0170