ARMmbed/mbed-os #4507: [Exporter] K66F not supported by IAR

yennster, 2017-06-08:
### Description
- Type: Bug
- Related issue: [#3633 [Exporter] [TLS-Client] : Platform not supported by IAR.](https://github.com/ARMmbed/mbed-os/issues/3633)
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target:**
K66F

**Toolchain:**
IAR

**Toolchain version:** 
7.8.0

**mbed-cli version:** 
1.1.1

**Expected behavior**
Export command should succeed.

**Actual behavior**
```
$ mbed export -i IAR -m K66F
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--source SOURCE_DIR]
                  [-D MACROS] [--profile PROFILE] [--update-packs]
project.py: error: K66F not supported by iar
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Repos\mbed-os-example-blinky\mbed-os\tools\project.py -i iar -m K66F --source ." in "C:\Repos\mbed-os-example-blinky"
```

**Steps to reproduce**
```
$ mbed import <any mbed-os-example-xxx>
$ mbed export -i IAR -m K66F
```

theotherjimmy, 2017-06-08:
> Export command should succeed.

My version of expected behavior:

The OOB matrix should not say to test this combo.

theotherjimmy, 2017-06-08:
also, you forgot the `cd <the mbed-os-example-xxx>` in your steps to reproduce :smile:

yennster, 2017-06-08:
@theotherjimmy I added clarification to the steps to reproduce section

mmahadevan108, 2017-07-31:
This should be fixed as part of https://github.com/ARMmbed/mbed-os/pull/4836

MarceloSalazar, 2017-09-14:
Verified it works with 5.6.0-rc1

closed 2017-09-14 by MarceloSalazar