ARMmbed/mbed-os #4512: Decimal part of both Humidity value and Pressure value are removed at BLE EnvironmentalService

soramame21, 2017-06-09:
It's an issue of BLE feature. 
BLE **EnvironmentalService** removed decimal part of both Humidity value and Pressure value before sending. only temperature value can be sent correctly with decimal part.
**EnvironmentalService** includes following 3 characterisitics.
1.  temperature
2.  humidity
3.  pressure

### Description
- Type: Bug
- Related issue: 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Tested Target**
TY51822r3,  it can be reproduced on any targets with a BLE shield.

**Expected behavior**
Both Humidity value and Pressure value are sent with decimal part.
**Actual behavior**
Decimal part of  Humidity and Pressure cannot be sent correctly. 
**Steps to reproduce**
Just calling API updateHumidity() and updatePressure()
**How to fixed**
Changed input value type from **HumidityType_t**, **PressureType_t** to **float**
Please look attached diff file.
[ren_diff0609.txt](https://github.com/ARMmbed/mbed-os/files/1063033/ren_diff0609.txt)

0xc0170, 2017-06-09:
cc @pan-  @ytsuboi @toyowata

soramame21, 2017-06-14:
@pan- This is an issue in our following header file. the input Humidity value and Pressure value are converted from float to integer (16/32 bit) before multiplying by 100/10, therefore decimal part is lost before sending out. the conversion to integer needs to be done after multiplication.
> /features/FEATURE_BLE/ble/services/EnvironmentalService.h

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .

0xc0170, 2021-08-13:
EnvironmentalService was deprecated, I'll close this as won't fix.

closed 2021-08-13 by 0xc0170