ARMmbed/mbed-os #4515: Undefined class if INTEE

Nodraak, 2017-06-09:
### Description
- Type: bug
- Related commit: 24de27c9895b167e07b5cd4ddaa585b6ca1b9864
- Priority: medium

Following commit 24de27c9895b167e07b5cd4ddaa585b6ca1b9864, mbed-os does not compile if the target does not support interrupts.

**Explanation:**

If the target does not support interrupts, `DEVICE_INTERRUPTIN` is not defined and the class `InterruptIn` in `drivers/InterruptIn.h` is not declared. In `drivers/UARTSerial.h:174`, an `InterruptIn` is used, which leads to an `error: 'InterruptIn' does not name a type` during compilation.

*Will submit PR soon*

0xc0170, 2017-06-09:
Thanks for reporting, will wait for the fix.

curiosity, target does not support interrupts?

Nodraak, 2017-06-09:
> curiosity, target does not support interrupts?

I am currently adding a new target for a new microcontroller (on behalf of the company I work for). But it's because I just started working, so I only have the most basic things, I will most probably implement interrupts soon :)

hasnainvirk, 2017-06-13:
Should there be a platform that doesn't support interrupts ? Mind boggling @Nodraak  :) Half baked ports do not tend to get in mbed-os. Probably you can keep this change in your local tree or otherwise as @0xc0170 decides.

Nodraak, 2017-07-03:
#4538 is merged, closing.

closed 2017-07-03 by Nodraak