ARMmbed/mbed-os #4527: FATFileSystem.cpp  int FATFileSystem::format   have  bug

wjffsx, 2017-06-12:
int FATFileSystem::format(BlockDevice *bd, int allocation_unit) {
    FATFileSystem fs;   //this is a not init variable
    int err = fs.mount(bd, false);
    if (err) {
        return err;
    }

FATFileSystem fs;  before use Variables are not initialized

0xc0170, 2017-06-12:
cc @geky

geky, 2017-06-12:
In C++, declaring a variable calls its constructor. The FATFileSystem constructor is here:
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L235-L240

The `fs.mount` function sets up the FATFileSystem to use the BlockDevice in the f_format function.

0xc0170, 2017-06-12:
@wjffsx have you had any problems with the function ? any runtime error/compile warning or?

wjffsx, 2017-06-13:
complie is ok ; in runtime ;i can not format SD card

geky, 2017-06-13:
Ah ok, then a few questions:
1. Do any asserts fail? (an error would be printed in debug mode)
2. What is the return value of the format function? It should return an error code.
3. Do you have a code snippet we could use to reproduce the failure?

wjffsx, 2017-06-16:
have found reason, code is ok;
before format() ; should  call unmount(); 

Use the following order:
unmount();
format();
mount();

closed 2017-06-16 by wjffsx

geky, 2017-06-16:
Ah thats good to know. Glad you found a fix.