ARMmbed/mbed-os #4528: K82F - Uart not working on master branch

deepikabhavnani, 2017-06-12:
### Description
- Type: Bug 
- Priority: Blocker  Minor

**Target**
K82F

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
Should get prints on serial console

**Actual behavior**
No prints on UART

**Steps to reproduce**
Add print statement in blinky example and build for K82F target.

Debugging this further, it seemed the clock source is not set properly for K82F. 
@thinkberg: Inside CLOCK_SetLpuartSrc() was removed from serial_init() and moved to clock init. But I don't see  CLOCK_SetLpuartSrc() being called from TARGET_K82F\TARGET_FRDM\fsl_clock_config.c

Setting clock source properly resolves this issue.

thinkberg, 2017-06-13:
It should be properly set in serial_api.c.
The issue is that setting the serial clock source in fsl_clock_config.c ignores the fact, that it can be reprogrammed. If you do switch to a different power mode, it is possible that the clock source changes. My changes in serial_api.c: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K82F/serial_api.c#L53 ensure the correct clock source is selected when running serial_init().

It did work when I committed it.

thinkberg, 2017-06-13:
Not quite right, the issue was that before it was always clock source 2U, which does not work when not using an external reference clock. Normally the clock source should be set in fsl_clock)_config as in our UBRIDGE config. I guess adding CLOCK_SetLpuartSrc(2U) in fsl_clock_config.c should fix this issue.

deepikabhavnani, 2017-06-13:
Since FRDM_K82F board normally uses the ext. ref clock, should we set the default clock source to 2U in serial_api.c?

thinkberg, 2017-06-13:
If a user decides to not use the ext ref. clock with a K82F it will break the UART. It makes total sense to put this in fsl_clock_config.c. The code in serial_api.c is currently written, so it identifies the clock source used and retrieves the frequency accordingly. However, you could use the fallback to 2U in https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K82F/serial_api.c#L69

thinkberg, 2017-06-13:
@deepikabhavnani: I guess that was what you were talking about? Because originally there was only a single line setting 2U as the clock source without the case statement.

mmahadevan108, 2017-06-13:
@deepakvenugopal Thanks for finding this issue. I had not tested K82F for sometime.

@thinkberg Thanks for updating the serial_api.c file. At the I added this file it was written for the FRDM-K82F file. Apologies for this, very glad to note that you had added support for a new K82F based board. I am made modifications to the code to move serial initialization to the board layer. Can you please test on your board.

closed 2017-07-11 by deepikabhavnani