ARMmbed/mbed-os #4534: xDot bootloader

chrissnow, 2017-06-12:
### Description
- Type:  Question
---------------------------------------------------------------

I'm trying to implement a bootloader for the xDot (STM32L151CC) and am struggling.

I have followed instructions here
https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/flash/
to implement the flash IAP routines but the bootloader hard faults while trying to program a page.

here are my fork's
https://github.com/chrissnow/mbed-os/tree/xDot-bootloader-5.47
https://github.com/chrissnow/mbed-os-example-bootloader/tree/xDot
https://github.com/chrissnow/mbed-os-example-bootloader-blinky/tree/xDot

the application also does not run if I program with the combined binary.

I tried to use mbed-os master but had all sorts of runtime errors relating to mutexes in ISR's despite being in the main routine.

cc @janjongboom

theotherjimmy, 2017-06-13:
@chrissnow You will have to submit any PRs against the master branch. Therefore, it's my recommendation that you try to work with the master branch from the start.

@c1728p9 both Flash IAP and mutex in ISR were your changes. Do you have any debugging or development guidelines that could help with implementing this?

chrissnow, 2017-06-13:
@theotherjimmy I did the changes against master but then back ported them to a 5.4.7 after issues.

@c1728p9 the mutex in ISR might not actually a problem as it seems to be working now against master at least for the DataFlash variant, I still have issues with the SD card but that's probably something else as different SD cards have different issues!

my master branch is here, examples have been updated to this branch.
https://github.com/chrissnow/mbed-os/tree/xDot-Bootloader

janjongboom, 2017-06-14:
@LiyouZhou @marcuschangarm Can mbed-os-example-bootloader be changed for the update-aware bootloader? We want to run update client on xDot.

chrissnow, 2017-06-15:
Thanks to a suggestion from @mfiore02 I now have a working bootloader :-)
rather than the CMSIS flash algorithm I am using the STM HAL such as
HAL_FLASH_Program
HAL_FLASHEx_Erase

It can either bootload from SD or DataFlash so perfect for development and production.

I will tidy it up and submit a PR soon.

mfiore02, 2017-06-15:
@chrissnow Nice! I just posted a ton of info and some code to your case on our support portal. Might be useless now but it's there anyway :smile_cat: 
Glad you were able to get everything working!
Cheers!

chrissnow, 2017-06-15:
All my forks have now been updated and should work.

Some info that might be of use for anyone wishing to give it a try....

- First 64k reserved for the bootloader, it is possible to reduce this but needs some work, 32k may be achievable.

- STM HAL API is missing the mbed flash api for L1, I took the L0 and tried to modify it without success but it looks to be something simple to fix, then my simplified apply_update can be reverted back the the original one, assistance from STM would be great on this...

- You will need to setup your SPI bus to your own hardware, unless you use the same pin's as me, you will want to update the bootloader bin in the app with your own (xDot.bin) if you want to combine it for drag and drop.

- if you want to use DataFlash instead of SD you will need to swap the #define, bear in mind you will need to format it and have a method of getting the .bin on there to test!

- mbed will generate two bin files, mbed-os-example-bootloader-blinky.bin is the combined app and bootloader, use this for drag and drop programming, mbed-os-example-bootloader-blinky_application.bin is what you need to copy to the SD card but must be renamed app.bin.

- better error checking needs implementing to prevent a failed flash

chrissnow, 2017-06-16:
@janjongboom  I was using ARMCC so didn't come across that, just testing now.
the info came from
https://docs.mbed.com/docs/mbed-os-handbook/en/5.4/advanced/flash/

janjongboom, 2017-06-16:
@chrissnow Ah, ok. I can use ARMCC.

chrissnow, 2017-06-16:
@janjongboom GCC_ARM should now work, it looks like the example was wrong or different for this MCU.
GCC doesn't fit in the region though currently so unable to test fully.
The branch has been updated.

janjongboom, 2017-06-19:
@chrissnow Good next step would be compiling w/o RTOS. One of your drivers depends on Mutex though, so that would need to be rewritten unfortunately. You can replace the Mutex'es with [PlatformMutex](https://github.com/ARMmbed/mbed-os/blob/master/platform/PlatformMutex.h) which is not RTOS dependent.

Then building with `NDEBUG=1` and without using printf should send it down under 32K.

chrissnow, 2017-06-23:
@janjongboom I have just pushed some changes, it now (just) fits in 36k if you compile with --profile mbed-os\tools\profiles\release.json

I'm not sure where to go next to reduce it further, I think it's too tight so would likely need to go to 40k, DataFlash isnt so tight with 0x494 of padding.

```Merging Regions:
  Filling region application with .\BUILD\xdot_l151cc\arm\mbed-os-example-bootloader_application.bin
  Padding region application with 0x6c bytes
Space used after regions merged: 0x9000
+---------------------+-------+-------+------+
| Module              | .text | .data | .bss |
+---------------------+-------+-------+------+
| Misc                | 10163 |    20 |  676 |
| drivers             |   472 |    12 |    0 |
| features/filesystem | 11412 |    12 |  544 |
| hal                 |   540 |     0 |   24 |
| platform            |  2487 |    53 |  324 |
| targets/TARGET_STM  | 11481 |    24 |  480 |
| Subtotals           | 36555 |   121 | 2048 |
+---------------------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 2169 bytes
Total RAM memory (data + bss + heap + stack): 2169 bytes
Total Flash memory (text + data + misc): 36676 bytes```

LMESTM, 2017-06-27:
@chrissnow  thanks for pushing this support. I'll make a short review of it.
Could you run the test cases related to flash and share the results ? 
tests-mbed_drivers-flashiap
tests-mbed_hal-flash

closed 2017-07-17 by chrissnow