ARMmbed/mbed-os #4535: [OOB 5.5.0] NUCLEO_F767ZI not supported by IAR exporter

toyowata, 2017-06-13:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target:**
NUCLEO_F767ZI

**Toolchain:**
IAR

**Toolchain version:** 
7.8.0

**mbed-cli version:** 
1.1.1

**Expected behavior**
Export command should succeed.

**Actual behavior**
```
$ mbed export -m NUCLEO_F767ZI -i iar
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--source SOURCE_DIR]
                  [-D MACROS] [--profile PROFILE] [--update-packs]
project.py: error: NUCLEO_F767ZI not supported by iar
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u E:\oob\mbed-os-example-blinky\mbed-os\tools\project.py -i iar -m NUCLEO_F767ZI --source ." in "E:\oob\mbed-os-example-blinky"
```

**Steps to reproduce**
```
$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed update mbed-os-5.5.0-rc1-oob
$ mbed export -m NUCLEO_F767ZI -i iar
```

theotherjimmy, 2017-06-13:
@toyowata This is not a bug. You are requesting an enhancement.

theotherjimmy, 2017-06-13:
Also, as this is true on master, I'm removing the OOB tag

0xc0170, 2017-06-14:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-06-14:
I am on it

toyowata, 2017-07-19:
Fixed in latest version.

closed 2017-07-19 by toyowata