ARMmbed/mbed-os #4549: Atmel ATSAMD21-XPRO mbedOS 5.x support

ravihansa3000, 2017-06-14:
### Description
- Type: Enhancement
- Priority: Minor
----------------------------------------------------------------

## Enhancement

**Reason to enhance or problem with existing solution**
Board currently only supports mbedOS 2

**Suggested enhancement**
Update board support for mbedOS 5.x 

**Pros**
None of the ATSAM series boards currently support mbedOS 5.x. Atmel's ATSAMD21 is a popular MCU, used by many vendors for evaluation boards (Arduino Zero, Adafruit Feather M0, SparkFun SAMD21 Mini). Having mbedOS5 support will help the mbedOS community to expand its target audience.

vargham, 2017-06-16:
Is Atmel still interested in further developing mbed compatible devices and drivers?

0xc0170, 2017-06-16:
cc @Parthasarathy @kpurusho @akhilpanayamparambil

helmut64, 2017-06-25:
I have upcoming projects with the D21 MCU, latest mbed support would be appreciated.

malachib, 2017-10-13:
I am very much interested in this also as per #3370

malachib, 2017-10-13:
@helmut64 similar boat here.  Arduino is great for what it is, but I am more confident in mbed for mission critical

akhilpanayamparambil, 2018-02-21:
@Parthasarathy are you guys planning to take this up ? I can help when I have time. :-)

helmut64, 2018-02-21:
@Parthasarathy SAMD21 is dead on mbed, also the D21 hardware is way behind compared to other MCUs. I moved on and spend my time to enhance active targets, e.g. STM32 in particular STM32L4.
This is my personal opinion.

malachib, 2018-02-22:
@helmut64 I can't disagree with your observations, but I still wish SAMD21 were supported.  I'd go so far as to say many non-hobbyists would also be interested in such a thing

Parthasarathy, 2018-02-22:
Hi,

   Please contact Microchip Support to know more details of mbed support
for SAMD21.
    http://www.microchip.com/support/hottopics.aspx

Thanks,
Partha

tuupola, 2018-02-23:
Oh, great. I had already chosen ATSAMD21 and mbedOS for a hobby project of mine. Did not notice before that mbedOS 5 is not supported.

tuupola, 2018-02-24:
If anyone is interested, answer from Microchip PR department is.

"Microchip does not currently support mbed OS."

This of course does not stop open source contributors working on it. Given the lack of support from manufacturer it might not be worth it.

helmut64, 2018-02-24:
The owner of this issue can close this case. @ravihansa3000

closed 2018-02-26 by ravihansa3000