ARMmbed/mbed-os #4564: cant debug threads on mbedos 5.5

orenc17A, 2017-06-15:
Hi
i'm debuging threads with eclipse and pyocd on the mbedos-5.5-rc2 tag
and it seems that eclipse cant tell the difference between the different threads
i'm currently debuging on K64F compiled with GCC_ARM 6

this wasn't a problem with 5.4.x

Thanks

pan-, 2017-06-15:
@c1728p9

c1728p9, 2017-06-15:
Hi @orenc17A, pyOCD + eclipse does not support task aware debugging. You can only see the callstack of the current thread.

You mention that this was working for you. Were you using some plugin with this functionality?

orenc17A, 2017-06-15:
nope, i installed eclipse and pyOCD according to [this guide](https://developer.mbed.org/users/c1728p9/notebook/debugging-with-eclipse-and-pyocd/)

c1728p9, 2017-06-16:
Hi @orenc17A those instructions don't have steps for task aware debugging. Can you attach a screenshot of this working with 5.4.x so I can see what your referring to?

closed 2018-06-02 by orenc17