ARMmbed/mbed-os #4570: Heap tracking in mbed-os

RobMeades, 2017-06-15:
## Question

I'm trying to see if I can fit a build into constrained RAM (mbedClient into C027).  I can reduce the static RAM a long way but "Operator new out of memory" is not helping me very much with the dynamic stuff.  Is there anything I can switch on in `mbed-os` to give me heap diagnostics of some form?

vargham, 2017-06-16:
Yes, this is possible: https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/runtime_stats/

0xc0170, 2017-06-16:
Does that answer your question @RobMeades ?

RobMeades, 2017-06-16:
Sort of: what I was hoping for was that, when operator new runs out of memory, it might say how much was asked for, how much was available, and give me a dump of what it knows about the heap.  Or it might allow me to provide a static buffer (I have RAM available that could not be `malloc()`ed from anyway) into which it would log the `malloc()`s as they occur.  But in the absence of that I'll add `mbed_stats_heap_get()` calls through the code that I can see; the problem will be if lots of the `malloc()`s are deep inside mbedClient or the like.

0xc0170, 2017-06-16:
It is currently not available. We can add hooks for allocations, to be able to add logging functionality. This could be a feature request.

RobMeades, 2017-06-16:
No problem, I can manage for now, pretty easy to add for oneself.

closed 2017-06-16 by RobMeades

RobMeades, 2017-06-16:
Actually, no, a hook is required, I just can't get enough information after the fact.  I'll raise a feature request.