ARMmbed/mbed-os #4575: Semaphore release occurs before acquisition for TCP socket over UART

mray19027, 2017-06-16:
Original issue raised which discovered the bug: https://github.com/ARMmbed/mbed-os-example-wifi/issues/34

PR addressing the bug: https://github.com/ARMmbed/mbed-os/pull/4560

## Replication steps

Example: https://github.com/ARMmbed/mbed-os-example-wifi
GCC_ARM version: `5.4.1 20160919 (release)`
mbed CLI version: `1.0.0`
mbed OS version: `92fbf2a9b3988d430482fc25a6077f2462e2a634`

```
WiFi example

Scan:
Network: AccessNG secured: Unknown BSSID: 70:3A:E:54:fe:80 RSSI: -67 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:54:fe:81 RSSI: -66 Ch: 1
Network: AccessNG secured: Unknown BSSID: 70:3A:E:61:d1:80 RSSI: -83 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:d1:81 RSSI: -87 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:55:7d:c1 RSSI: -72 Ch: 1
Network: ESP_1CCF60 secured: None BSSID: 5E:CF:7F:1c:cf:60 RSSI: -46 Ch: 6
Network: GORDON-AP secured: None BSSID: B8:27:EB:d1:f9:c1 RSSI: -52 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:c0:61 RSSI: -69 Ch: 6
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -45 Ch: 6
Network: ESP_DC83BA secured: None BSSID: 1A:FE:34:dc:83:ba RSSI: -58 Ch: 6
Network: bdn_wireless secured: None BSSID: 8A:15:4:ca:90:f0 RSSI: -85 Ch: 6
Network: Brandywine Public secured: None BSSID: 8A:15:4:ca:90:f6 RSSI: -85 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:61:c0:60 RSSI: -69 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:54:fc:60 RSSI: -81 Ch: 6
14 networks available.

Connecting...
Success

MAC: 18:fe:34:f9:e8:20
IP: 192.168.1.39
Netmask: 255.255.255.0
Gateway: 192.168.1.1
RSSI: -49

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20001E00, task name: (null))
```

kegilbert, 2017-06-28:
I'm seeing this behavior with the latest mbed-os (a39ac6030 ) on a K64F+ESP8266, and mbed-os-example-wifi commit hash of (8263817).

nelqatib, 2017-06-28:
@kegilbert Have you tried with the changes in [this](https://github.com/ARMmbed/mbed-os/pull/4560/) closed PR?

kegilbert, 2017-06-28:
@nelqatib I did, the error did not show up but the example hung.

kegilbert, 2017-06-30:
Recursive receive call in the ATParser was overrunning the main thread's stack space. @geky recommended the following fix due to an earlier PR to do this.

https://github.com/ARMmbed/ATParser/pull/5

kegilbert, 2017-07-07:
Is anyone still seeing this error?  This should be good to close if no one can reproduce, above fix went in earlier this week.

closed 2017-07-10 by 0xc0170