ARMmbed/mbed-os #4576: mbed tick invalid after waking from stop mode on xDot

mfiore02, 2017-06-16:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
XDOT_L151CC

**Toolchain:**
GCC_ARM|ARM
Did not test IAR but is likely an issue with that toolchain as well.

**Toolchain version:**
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
uVision V5.21.1.0

**mbed-cli version:** 
0.9.1

**meed-os sha:**
fc18365 Merge pull request #4463 from ARMmbed/release-candidate (mbed-os-5.4.7)

**DAPLink version:**
0241

**Expected behavior**
After waking from a low power mode, the us tick should function just like it did before entering a low power mode.

**Actual behavior**
The prescaler for TIM5 (the mbed us tick) has changed from 31 to 9, causing issues with Timers, Tickers, etc.

**Steps to reproduce**
The following application reproduces with mbed-os-5.4.7.

```c++
#include "mbed.h"
#include "hal_tick.h"
#include "mbed_rtx.h"

Serial pc(USBTX, NC);

void countdown(int count) {
    DigitalOut led(LED1);
    Timer timer;
    timer.start();

    while ( count-- > 0 ) {
        printf("Count %d: %d msec\r\n", count, timer.read_ms());
        led = 1; wait_ms(150);
        led = 0; wait_ms(850);
    }
}

void fix_tick() {
    if ( TIM_MST->PSC != OS_CLOCK / 1000000 - 1 ) {
        TIM_MST->PSC = OS_CLOCK / 1000000 - 1;
        TIM_MST->EGR = TIM_EVENTSOURCE_UPDATE;
        while ( TIM_MST->EGR & ~TIM_EVENTSOURCE_UPDATE ) {};
    }
}

int main() {
    pc.baud(115200);

    countdown(5);

    hal_deepsleep();
    //fix_tick();

    countdown(5);
}
```

Without the tick prescaler getting fixed, output will demonstrate that the tick has been broken.

> Count 4: 0 msec
> Count 3: 1001 msec
> Count 2: 2003 msec
> Count 1: 3004 msec
> Count 0: 4006 msec
> Count 4: 0 msec
> Count 3: 3202 msec
> Count 2: 6406 msec
> Count 1: 9609 msec
> Count 0: 12812 msec

If the tick prescaler is fixed using the fix_tick function, output will be as expected.

> Count 4: 0 msec                                                    
> Count 3: 1001 msec                                                 
> Count 2: 2003 msec                                                 
> Count 1: 3004 msec
> Count 0: 4006 msec
> Count 4: 0 msec
> Count 3: 1001 msec
> Count 2: 2003 msec
> Count 1: 3004 msec
> Count 0: 4006 msec

klaas019, 2017-10-24:
Works as of mbed-os-5.5.7

chrissnow, 2017-10-24:
Was it properly fixed or just the MTS workaround that was implemented?

klaas019, 2017-10-24:
It was properly fixed, not sure in what version though just as of 5.5.7.

0xc0170, 2017-10-24:
> It was properly fixed, not sure in what version though just as of 5.5.7.

It would be good if the patch referenced the issue, can you provide the number, we could close this as resolved.

klaas019, 2017-10-24:
I don't know what patch you're referring to. All I know is it didn't work in mbed-os-5.4.7, and now works in mbed-os-5.5.7 and praying to the mbed-os gods that it works in later versions as well.

0xc0170, 2017-10-24:
I would like to locate a pull request that fixed this bug (it is not referenced, so possibly some L1 patch did fixed this) . @mfiore02 Can you confirm?

I am currently going through the git log for L1 targets, and can't find the commit that would be related and possibly fixed.

chrissnow, 2017-10-24:
I also couldn't see anything that would have fixed it, MTS put a fix in their library so that on wake it hid the problem, There was a debug message each time it woke up to say so.

@0xc0170  I heard @mfiore02 is no longer at MTS so it might be worth tagging someone else too.

klaas019, 2017-10-24:
Yes @mfiore02 left MTS.

0xc0170, 2017-10-27:
resolved as confirmed above.

Thanks @klaas019

closed 2017-10-27 by 0xc0170