ARMmbed/mbed-os #4582: Major Bug: LPC15XX Extended CAN Read Not working

sivasuren1, 2017-06-17:
### Description
Bug-Major 

---------------------------------------------------------------
## Bug

**Target**
LPC15XX

**Toolchain:**
ARM
**Expected behavior**
When I issue a command in normal (CANStandard) I get a a response. However When I send by CAN Extended, the ISR fires but can.read returns 0. We tried the same code on LCP1768 and it works fine

Our CAN Similator shows that it responds the same way with boht LPC1768 and LCP1549 but the code in LPC1549 doesnt not read it

**Actual behavior**
CAN.READ() issues a 0 instead of. 

**Steps to reproduce**
Code

CAN can2(P0_22, P0_23);
Serial serialPort(USBTX, USBRX);
CANMessage msg;

char counter = 0;
char can_msg[8];
void setCanMessage(char* _cmd);

void setCanMessage(char* _cmd)
{
    can_msg[0] = 2;
    can_msg[1] = 1;
    can_msg[2] = 13;
    can_msg[3] = 0;
    can_msg[4] = 0;
    can_msg[5] = 0;
    can_msg[6] = 0;
    can_msg[7] = 0;
}

void send()
{
    setCanMessage("010D");

    can2.reset();
    can2.frequency(500000);

    if (can2.write(CANMessage(PID_REQUEST,can_msg,8,CANData,CANExtended))) {
        printf("Message sent\r\n");
    }
}

void canRead()
{
        CANMessage canMessage;
        can2.read(canMessage);
        serialPort.printf("%02X\r\n",canMessage.len);
        for (int i = 0; i < (int)canMessage.len; i++) {
            serialPort.printf("%02X",canMessage.data[i]);
        }
        char *message = (char *)"\r\nCAN RX IRQ\r\n\0";
        serialOut(message);
}

int main()
{
    can2.attach(canRead);
    while(1) {
        serialPort.printf("loop()\r\n");
        send();
        wait(2);
    }
}

0xc0170, 2017-06-19:
cc @toyowata

toyowata, 2017-08-10:
I think that the ```can_read()``` does not seem to handle new data correctly [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC15XX/can_api.c#L546):

```cpp
#define RX_MSG_OBJ_COUNT 31
// snip
int can_read(can_t *obj, CAN_Message *msg, int handle) {
// snip
for (i = 0; i < RX_MSG_OBJ_COUNT; i++) {
```

This should be:

```cpp
for (i = 0; i <= RX_MSG_OBJ_COUNT; i++) {
```

The current code does not handle MSB of the newdata.

0xc0170, 2018-06-19:
@sivasuren1 Hi, thanks for  the report. Have you resolved the issue? Can you send a new pull request with a fix to address this CAN issue?

ithinuel, 2021-02-17:
@donatieng @MarceloSalazar LPC15xx do not seem to be supported in mbed any more.
In a way, this issue is now fixed.

closed 2021-02-18 by MarceloSalazar