ARMmbed/mbed-os #4583: STM32F429ZI CAN2 transmit fails on PB_13

vargham, 2017-06-18:
CAN2 can not transmit on PB_13. All other variations are OK, both with CAN1 and with CAN2. Checked on a real CAN bus with logic analyzer.

### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
STM32F429ZI (Discovery board)

**Toolchain:**
GCC_ARM|ARM (EmBitz, Online compiler)

**Expected behavior**
Successful transmission on PB_13.

**Actual behavior**
CAN2 can not transmit on PB_13.

**Steps to reproduce**
Try to transmit a CAN frame with CAN interface 2 on PB_13 as CAN TD pin. See example code below.

```
#include "mbed.h" //5.4.7 (144)

const PinName can1rxPins[] = {PB_8, PD_0, PA_11};
const PinName can1txPins[] = {PB_9, PD_1, PA_12};
const PinName can2rxPins[] = {PB_12, PB_5};
const PinName can2txPins[] = {PB_13, PB_6};

//CAN1
//0,0=OK 0,1=OK 0,2=OK 1,0=OK 1,1=OK 1,2=OK 2,0=OK 2,1=OK 2,2=OK
//CAN2
//0,0=RX_OK 0,1=OK 1,0=RX_OK 1,1=OK

//CAN can(can1rxPins[0], can1txPins[0]);
CAN can(can2rxPins[0], can2txPins[0]);

Serial serialDebug(PA_9, PA_10);  //UART1, Tx, Rx (Debug)
DigitalOut led1(LED1);
DigitalOut led2(LED2);

Thread sendThread(osPriorityAboveNormal, 2048);

CANMessage messageIn;
CANMessage messageOut;

void canRxIsr()
{
    can.read(messageIn);
    led1 = !led1;
}

void sendMessage()
{
    int status = can.write(messageOut);
    serialDebug.printf("Send status: %d\r\n", status);
}

void sendMessageLoop()
{
    while (true)
    {
        sendMessage();
        osDelay(1000);
    }
}

int main()
{
    serialDebug.baud(460800);
    serialDebug.printf("CAN receive / transmit test.\r\n");
    can.frequency(125000);

    messageOut.id = 0xD70000E;
    messageOut.format = CANExtended;
    messageOut.len = 0;

    can.attach(&canRxIsr);
    sendThread.start(&sendMessageLoop);

    while (true)
    {
        led2 = !led2;
        led1 = 0;
        osDelay(1000);
    }
}
```

jeromecoutant, 2017-06-19:
Hi
Note that on DISCO-F429ZI, PB_13 is also connected to USB VBUS

bcostm, 2017-06-19:
There is also PB_12 pin used also by OTG_FS_ID. The PeripheralPins.c file need to be updated in order to remove all pins used by other boards components. We have done it for the recent boards but not on this one.

jeromecoutant, 2017-07-12:
Hi
For information, Bruno's work can be seen in #4668

jeromecoutant, 2017-07-12:
CAN loop test MBED_A28 is OK with

LMESTM, 2017-09-05:
#4668 has been merged - so should we close this issue ?

0xc0170, 2017-09-05:
Please reopen with new findings if the issue persist

closed 2017-09-05 by 0xc0170